ouster-lidar/ouster-sdk #334: [ouster python sdk] The difference between pcap image and bag image

JJuhyeon-Kim, 2022-01-10:
Hi, thank you for your hard work as always.

I have a question about the image acquired using LiDAR.
The images of pcap file are as follows.
[image]
I converted pcap images as gray scale, but they look a little different from the image in the bag file. (range, intensity, and ambient image respectively)
[image]
However, the images of bag file based on ROS are as follows. (range, intensity, and ambient image respectively)
[image]
I think the image acquired as a bag file went through a preprocessing process.
So, the ouster python sdk does not include the code for the preprocessing process yet?

If you answer, it will be of great help to me.
Thank you.

kairenw, 2022-02-03:
Hi @JJuhyeon-Kim,

That's correct. The image in the bag file has preprocessing applied [here](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/img_node.cpp#L153). If you want to apply something similar in Python SDK, you can use [this function](https://github.com/ouster-lidar/ouster_example/blob/master/python/src/ouster/sdk/examples/colormaps.py#L20) or something like it. It won't be exactly the same but it should produce a similar result. [Here's](https://github.com/ouster-lidar/ouster_example/blob/ad837d005e607e0ec1a16e967705cce15b0fb851/python/src/ouster/sdk/examples/pcap.py#L105) an example of it being used, so you can get some context.

That function and its uses within our codebase are currently all provided only as example code, and not part of our API/SDK, so they may be removed in the future. Therefore, I'm going to keep this ticket open as a 'feature request' for us to provide preoprocessing as part of the Python SDK. I think that should address your issue though. Thanks for asking, and please let us know if you have more questions.

SvenMuc, 2022-09-06:
Are there any concrete plans to add the `AutoExposure` and `BeamUniformityCorrector` C++ classes to the python SDK?
That would be really usefull.

SvenMuc, 2022-09-07:
Thanks for the quick reply, that helped a lot.

twslankard, 2023-08-14:
I'm closing this issue due to inactivity. Please open a new issue for SDK-specific inquiries or contact support@ouster.io for device-related support.

Best Regards,

Tom
Staff Software Engineer
Ouster, Inc.