ouster-lidar/ouster-sdk #336: Calling config service failed: 'std::invalid_argument' what():  n_cols_of_lidar_mode

emigomez, 2022-01-14:
Hi,

I'm having an error during the execution of the ROS project on Ubuntu20.
I have updated the version of firmware to the last one, and I think I have correctly configured the sensor_hostname and the udp_dest to receive the lidar data. However, this is the error that I am having:

[ INFO] [1642159495.739683641]: Client version: 0.3.0+13ea8e8-dirty-release
[ INFO] [1642159495.740330843]: Waiting for sensor 192.168.1.157 to initialize ...
[ INFO] [1642159495.740555730]: Sending UDP data to 192.168.1.111
process[relay_lidar-4]: started with pid [2807]
process[relay_imu-5]: started with pid [2817]
[ INFO] [1642159498.804909033]: Sensor configured successfully
[ INFO] [1642159501.876494114]: Sensor initialized successfully
[ INFO] [1642159501.876911754]: Wrote metadata to /home/emilio/Desktop/lidar_meta.json
WARNING: No data_format found.
terminate called after throwing an instance of 'std::invalid_argument'
  what():  n_cols_of_lidar_mode
[ERROR] [1642159502.079350421]: Calling config service failed
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 2804, exit code -6, cmd /home/emilio/uav-sensors/ouster-ros-ws/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/emilio/.ros/log/99263e72-752c-11ec-9372-f1f21801c130/os_node-2.log].
log file: /home/emilio/.ros/log/99263e72-752c-11ec-9372-f1f21801c130/os_node-2*.log
Initiating shutdown!
================================================================================
[relay_imu-5] killing on exit
[relay_lidar-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I have tried with different paths for the metadata, and also with different lidar_mode as the error appear to be related to these parameters.

This is my sensor configuration in the ouster.launch file:
 <arg name="sensor_hostname" default="XXXXXXXXXX" doc="hostname or IP in dotted decimal form of the sensor"/>
  <arg name="udp_dest" default="XXXXXXXXXX" doc="hostname or IP where the sensor will send data packets"/>
  <arg name="lidar_port" default="7502" doc="port to which the sensor should send lidar data"/>
  <arg name="imu_port" default="7503" doc="port to which the sensor should send imu data"/>
  <arg name="replay" default="false" doc="do not connect to a sensor; expect /os_node/{lidar,imu}_packets from replay"/>
  <arg name="lidar_mode" default="1024x10" doc="resolution and rate: either 512x10, 512x20, 1024x10, 1024x20, or 2048x10"/>
  <arg name="timestamp_mode" default="" doc="method used to timestamp measurements: TIME_FROM_INTERNAL_OSC, TIME_FROM_SYNC_PULSE_IN, TIME_FROM_PTP_1588"/>
  <arg name="metadata" default="XXXXXXXXXX/lidar_meta.json" doc="path to read or write metadata file when replaying or receiving sensor data, respectively"/>
  <arg name="viz" default="false" doc="whether to run a rviz"/>
  <arg name="rviz_config" default="-d $(find ouster_ros)/viz.rviz" doc="optional rviz config file"/>
  <arg name="tf_prefix" default="os1" doc="namespace for tf transforms"/>
  <arg name="udp_profile_lidar" default="" doc="lidar packet profile: LEGACY, RNG19_RFL8_SIG16_NIR16_DUAL"/>

What could be the error?
Thanks in advance,

emigomez, 2022-01-14:
Problem solved following all the steps of the repo.
I just was trying to compile the ROS project. Following all the steps of the repo, the execution works and it is able to check all the parameters as the lidar_mode inside the lidar_client project

kairenw, 2022-01-14:
Whoops, the errors on the node are not clear right now. This error likely happened because your sensor actually could not be reached at that IP. Sorry about the confusion, we will fix that up for the next push.

Closing this issue now as it seems it's resolved for you.

closed 2022-01-14 by kairenw