ouster-lidar/ouster-sdk #339: Calling config service failed

krjones147, 2022-01-28:
I'm running this as my roslaunch command
roslaunch ouster_ros ouster.launch sensor_hostname:=os991931000518 metadata:=OS-1-991931000518-1024x10.json lidar_mode:=1024x10 viz:=true
This is the error that I'm getting
[ERROR] [1643391609.679433142]: Calling config service failed
[ERROR] [1643391609.679433044]: Calling os config service failed
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 27789, exit code -6, cmd /home/catlens/ousterWorkspace/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/catlens/.ros/log/56167404-8061-11ec-806f-ffa56ce096f2/os_node-2.log].
log file: /home/catlens/.ros/log/56167404-8061-11ec-806f-ffa56ce096f2/os_node-2*.log
Initiating shutdown!
================================================================================

kairenw, 2022-01-28:
Hi @krjones147,

Could you paste in the entire log from where you `roslaunch` ?

Thanks!

krjones147, 2022-01-28:
Hi @kairenw ,
This is the entire log

roslaunch ouster_ros ouster.launch sensor_hostname:=os991931000518 metadata:=OS-1-991931000518-1024x10.json lidar_mode:=1024x10 viz:=true
... logging to /home/catlens/.ros/log/1a09af02-8062-11ec-806f-ffa56ce096f2/roslaunch-ECE-13657-L-28123.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ECE-13657-L:35731/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 1024x10
 * /os_node/lidar_port: 0
 * /os_node/metadata: OS-1-991931000518...
 * /os_node/replay: False
 * /os_node/sensor_hostname: os991931000518
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 
 * /os_node/udp_profile_lidar: 
 * /rosdistro: noetic
 * /rosversion: 1.15.13

NODES
  /
    img_node (ouster_ros/img_node)
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [28131]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1a09af02-8062-11ec-806f-ffa56ce096f2
process[rosout-1]: started with pid [28141]
started core service [/rosout]
process[os_node-2]: started with pid [28148]
process[os_cloud_node-3]: started with pid [28149]
process[rviz-4]: started with pid [28150]
process[img_node-5]: started with pid [28155]
process[relay_lidar-6]: started with pid [28161]
[ INFO] [1643391938.205877443]: Client version: 0.3.0+unknown-release
[ INFO] [1643391938.207345675]: Waiting for sensor os991931000518 to initialize ...
[ INFO] [1643391938.207579194]: Using automatic UDP destination
process[relay_imu-7]: started with pid [28163]
cfg getaddrinfo(): Temporary failure in name resolution
[ INFO] [1643391938.215963398]: Sensor configured successfully
cfg getaddrinfo(): Temporary failure in name resolution
[ INFO] [1643391938.222654447]: Sensor initialized successfully
[ INFO] [1643391938.223008267]: Wrote metadata to OS-1-991931000518-1024x10.json
WARNING: No data_format found.
terminate called after throwing an instance of 'std::invalid_argument'
  what():  n_cols_of_lidar_mode
[ INFO] [1643391938.346112646]: rviz version 1.14.10
[ INFO] [1643391938.346150602]: compiled against Qt version 5.12.8
[ INFO] [1643391938.346160083]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1643391938.353620706]: Forcing OpenGl version 0.
[ERROR] [1643391938.390638004]: Calling os config service failed
[ERROR] [1643391938.390638173]: Calling config service failed
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 28148, exit code -6, cmd /home/catlens/ousterWorkspace/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/catlens/.ros/log/1a09af02-8062-11ec-806f-ffa56ce096f2/os_node-2.log].
log file: /home/catlens/.ros/log/1a09af02-8062-11ec-806f-ffa56ce096f2/os_node-2*.log
Initiating shutdown!
================================================================================
[ INFO] [1643391938.538896247]: Stereo is NOT SUPPORTED
[ INFO] [1643391938.538954148]: OpenGL device: Mesa Intel(R) UHD Graphics (CML GT2)
[ INFO] [1643391938.538973341]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
[relay_imu-7] killing on exit
[relay_lidar-6] killing on exit
[img_node-5] killing on exit
[rviz-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kairenw, 2022-01-28:
Yeah, unfortunately in the latest push we wrote the errors pretty unclearly, which is basically entirely my fault since I'm the one who updated those lines, lol. I think it's likely your sensor can't be reached at the current hostname you're providing. You can see it on this line:
```cfg getaddrinfo(): Temporary failure in name resolution```

Try pinging `os991931000518`, and if that doesn't work then that's the reason. You probably want something like `os-991931000518.local` (though I'm just guessing).

krjones147, 2022-01-28:
Pinging os-991931000518 returned ping: os-991931000518: Temporary failure in name resolution
I tried adding .local and pinging it and got this: ping: os-991931000518.local: Name or service not known

kairenw, 2022-01-28:
Not to worry! It's acutally not ROS, we're just trying to find the hostname of your sensor. Are you sure you've got the connection right? 

Can you follow the connection steps on page 129 of [our sensor manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2.2.x.pdf) again?

kairenw, 2022-01-28:
a-ha! is it possible your hostname is actually `os1-991931000518.local` ? Try throwing that in the mix after you recheck the connection

krjones147, 2022-01-28:
So you are correct that I had the wrong host name. I confirmed that using avahi-browse -lrt _roger._tcp 
Unfortunately I'm still having issues connecting to the sensor. When I ping the correct host name I still get name or service not known

kairenw, 2022-01-28:
Can you reach the sensor hostpage? Try using that hostname you find in avahi-browse: http://<HOSTNAME>

krjones147, 2022-01-28:
I'm not getting anything to come up when I do that as well

kairenw, 2022-01-28:
If you don't mind, I'm going to send you to [our service portal.](https://ouster.atlassian.net/servicedesk/customer/portal/8) They can jump on a call and help you connect the sensor properly.

You can reference this ticket here.

krjones147, 2022-01-28:
I don't know what changed but it is working now. Thank you for your help

kairenw, 2022-01-28:
Oh, excellent! I'm so glad. Will close this ticket now.

closed 2022-01-28 by kairenw