ouster-lidar/ouster-sdk #340: After changing UDP Profile, there are no data

Shibaditya99, 2022-02-01:
I'm using OS2 Rev-06 model with v2.2.0 firmware. When I'm trying to test the Dual Return feature by changing the UDP Profile. After changing the profile there are no data. The diagnostics message is below.
    `Client machine announced it is not reachable on the provided not reachable on IMU data port; check that udp_dest and udp_port_imu configured on the sensor matches client IP and port.`

But when I'm changing the UDP Profie to "Legacy", It's showing data even with ouster studio.
As per the diagnostic message I think it's a wrong IMU port, But here I'm using the default port number.
So,  is there any solution for it?
And if I want to change the UDP Profile configuration with python, then what is the command for it?

kairenw, 2022-02-01:
This example should show how to change the UDP Profile configuration with Python: https://static.ouster.dev/sdk-docs/examples.html#configuring-your-sensor

What you're observing is quite strange. Would you mind contacting us through [our support portal](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86)? It'd also be nice if you could provide a log and/or a screenshot/video of the observed behavior.

Shibaditya99, 2022-02-02:
Okay

kairenw, 2022-02-03:
I will close this now, please let feel free to open it if you have any more questions / support doesn't get to you.

closed 2022-02-03 by kairenw