ouster-lidar/ouster-sdk #341: is there is a way to set lidar_to_sensor_transform?

hichamhendy, 2022-02-02:
I mus mount the LiDAR with angle 180 deg. or maybe another angle. Anyway I think I only need to set the intrinsic myself I tried to set it in metadata json file: However it changed, which proves no real usability for me

I also tried to pass in the transformation matrix while launching on ROS but nothing!
Does anybody have an idea?

kairenw, 2022-02-03:
Hi, @hichamhendy, 

Are you looking to change `lidar_to_sensor_transform` so that the `make_xyz_lut` returns points in the frame you desire? You can do so by replacing [this call to make_xyz_lut](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os_cloud_node.cpp#L71) with [one where you pass in the transform yourself](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/lidar_scan.h#L249). Just remember that you transform should be the matrix multiplicative output of info.lidar_to_sensor_transform and the transformation on top of that that you watn to apply.

You can then also modify `os_cloud_node` to publish your transform

Hope that helps!

Closing this out, but feel free to open it again if you have more questions.

closed 2022-02-03 by kairenw

hichamhendy, 2022-02-03:
Thanks you, sir! Your're answer is just perfect! I implemented it and it went fine. The reflective image is still overturned since I only flipped the transformed the LiDAR points. However, it's not relevant for me at the time being. In case it becomes, I would comment here.

Thank you again