ouster-lidar/ouster-sdk #343: Intermittent inability to read sensor

trevorki, 2022-02-07:
I'm trying to read data from my OS-1 using the python sdk on windows 11. To read a single scan I am using the example code as follows:
```
source = client.Sensor(hostname)
with closing(client.Scans(source)) as scans:
    scan = nth(scans, 1)
```

Sometimes this works successfully, but very often throws an error: **ClientTimeout: No packets received within 1.0s** and can't resume reading the from the sensor. I suspect the sensor  ports might be left open so I have tried throwing in `source.close()`, and have tried rebooting everything but still no dice.
The firewall is off and pinging the device works fine. The same thing occurs when trying to recording pcap files.

Any ideas how to resolve this?

kairenw, 2022-02-28:
Hi @trevorki,

Sorry for the long delay in response. Which version of the firmware are you on? Are you on 2.2 by any chance?

trevorki, 2022-02-28:
Yes indeed, 2.2.0

kairenw, 2022-03-07:
Have you tried looking at the networking traffic? You could use something like wireshark. Something like [`Get-NetAdapterStatistics`](https://docs.microsoft.com/en-us/powershell/module/netadapter/get-netadapterstatistics?view=windowsserver2022-ps) will also help us figure out what's going on.

Unfortunately, this error is a symptom of many different types of networking issues, so it's a bit of a hunt.

kairenw, 2022-05-23:
Hey @trevorki, let us know if you're still having issues. I'll close this out for now.

closed 2022-05-23 by kairenw