ouster-lidar/ouster-sdk #344: Deprecated functions should have comments on what's the replacement.

versatran01, 2022-02-12:
Otherwise it is difficult to migrate client code depending on older version.

kairenw, 2022-02-12:
Hi @versatran01,

Sorry about the difficulty. For `col_frame_id` we recommend using [`frame_id`](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/types.h#L464) instead. 

The encoder count is has been deprecated in our firmware because it is redundant with `measurement_id`. Here is the relevant explanation from the LEGACY Data Packet Format explanation in our [sensor user manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2.2.x.pdf#300) (page 18): 

```
Measurement ID [16 bit unsigned int] - a sequentially incrementing measurement counting up from 0 to
511, or 0 to 1023, or 0 to 2047 depending on lidar_mode.
[...]
Encoder Count [32 bit unsigned int] - an azimuth angle as a raw encoder count, starting from 0 with a max
value of 90,111 - incrementing 44 ticks every azimuth angle in 2048 mode, 88 ticks in 1024 mode, and 176
ticks in 512 mode
```
Thus, where you are currently using `col_encoder` we encourage you to use instead [`col_measurement_id`](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/types.h#L494), keeping in mind that you should multiply as necessary depending on mode.

If you have any other questions regarding recommended replacements of deprecated functions, please don't hesitate to ask.

I'll leave this open and mark it as "enhancement" so we remember to update the comments next time we deprecate functions.

versatran01, 2022-02-13:
Thanks for the answer. One quick question, how does one get the ticks per col_measurement_id? There is encoder_ticks_per_rev in packet_format, and I assume one can just divide that by the number of columns?

kairenw, 2022-02-14:
It's hard for me to see why you would choose to work with the ticks as opposed to editing the code to work with measurement_id, but if you need to, you can simply multiply based on the mode. That is, according to the definition cited above from the user manual, if you are in 2048 mode, multiply the measurement id by 44 ticks. In 1024 mode, by 88 ticks; and in 512 mode, by 176 ticks.

I would recommend that you refactor your code to use the measurement_id purely instead of relying on converting to ticks.

closed 2022-02-14 by kairenw