ouster-lidar/ouster-sdk #345: How can I get the azimuth and altitude angle of a single point?

Shibaditya99, 2022-02-14:
Hi,  I need to know that when I'm using the _beam_azimuth_angles_ or _beam_altitude_angles_ api like..
```
AZ_angle = stream.metadata.beam_azimuth_angles 
AL_angle = stream.metadata.beam_altitude_angles
``` 
then I'm getting a list of 128 values.
I also tried the equation which is mentioned on the software manual i.e.
`Theta_azimuth = -2*math.pi * (AZ_angle[i]/360)
Phi_altitude = 2*math.pi*(AL_angle[i]/360)`
So with these method I'm getting a value.
I'm using the stream method to display data. So I want to get the actual azimuth and altitude value of a single point in the data stream. So is that possible with the previous method? Or is there any other method is available to finding the value of the azimuth and altitude for a single point/pixel?

kairenw, 2022-02-14:
Hi @Shibaditya99, 

Thanks for the questions. It looks like you're working in Python. If you aren't, please let me know.

If you have a 128-beam sensor, then you should get 128 values from your `beam_azimuth` and `beam_altitude` angles, so that checks out. For a given point, the `i` in the equation you've written there should be the beam index starting from the top (and 0-indexed).

What are you trying to do more generally? You can already get all the xyz points by using our `XYZLut`: Try following the instructions here: [https://static.ouster.dev/sdk-docs/examples.html#projecting-into-cartesian-coordinates](https://static.ouster.dev/sdk-docs/examples.html#projecting-into-cartesian-coordinates). The output will be an array of `H x W x 3` doubles, so to obtain the `x`, `y`, and `z` coordinates from a `scan` of say, the point from the 17th and 1059th column, you would have:

```
xyzlut = client.XYZLut(info)
xyz = xyzlut(scan)
x = xyz[16, 1058, 0]
y = xyz[16, 1058, 1],
z = xyz[16, 1058, 2]
```

Hope that helps!

Shibaditya99, 2022-02-15:
Thanks for the reply @kairenw , 
`with closing(client.Scans.stream(hostname, lidar_port,
                                 complete=False)) as stream:
    show = True
    while show:
        for scan in stream:
            range = client.destagger(stream.metadata,
                                      scan.field(client.ChanField.RANGE))
            rng = (range[:, 1400:1900] / np.max(range) * 255).astype(np.uint8)`

Like this, I'm taking the stream data and I can select a pixel and get the RANGE data
`pixel_x = [26, 27]
 pixel_y = [370, 371]
X_slice= slice(pixel_x[0], pixel_x[1])
Y_slice = slice(pixel_y[0], pixel_y[1])
pixel_rng = rng[Y_slice, X_slice]
pixel_rng = pixel_rng.astype(np.uint32)*np.max(range)//255`

now I want to get the angles of that pixel. Is it possible this way?

kairenw, 2022-02-15:
Given a specific row and column of a pixel in a destaggered image, it is possible to get the altitude and azimuth angles associated. The method is a bit complex right now due to some complexities from the destaggering. Could you by any chance attach your metadata file?

(By the way, are you running that second slice of code on `range` or `rng`? `rng` is resized so that I am not quite sure how the pixels would be associated).

 If you're not comfortable posting it publically, you can reach me [through Ouster support ](https://ouster.atlassian.net/servicedesk/customer/portal/8). Just ask to put in contact with me and link this issue, please.

Shibaditya99, 2022-02-16:
@kairenw  Yes, I did use the `rng`, Then I'm reconverting the values. Yes it's complex.  actually I wanted to measure the Height and Width of a perticular object from the front using that 2D streaming. Is there any another way to get the measurement?
Just like the scale tool of Ouster studio. So they're using some algorithym for sure.

kairenw, 2022-02-20:
Hi @Shibaditya99, did you ever contact our user support? I can help you figure out if I have the sensor metadata for the specific mode you're interested in.

closed 2022-02-25 by Shibaditya99