ouster-lidar/ouster-sdk #351: [Help] How to use ROS timestamp to publish cloud and imu topics

TurtleZhong, 2022-02-24:
Hi, I wanna use ouster with other sensors like realsense cameras, So I need use the ros timestamp to publish pointclouds and imu topics, I followed #74 but still failed, below is my terminal output

Ouster Sensor: OS1-64
firmware: v2.1.2

my hostpc:

```sh
enp89s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.5  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::c2f5:9cd3:4ee2:c4f6  prefixlen 64  scopeid 0x20<link>
        ether 48:21:0b:25:b7:cf  txqueuelen 1000  (Ethernet)
        RX packets 14007324  bytes 19948899717 (19.9 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3604  bytes 673982 (673.9 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0x6c200000-6c2fffff 
```

and I check the network 

```sh
sudo ethtool -T enp89s0 
Time stamping parameters for enp89s0:
Capabilities:
	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
```
it means that the card only support software ptp sync?

Then I tried to turn timestamp_mode into TIME_FROM_PTP_1588 in launch file, I checked the header of the topic, it just like below:
```sh
stamp:
  secs: 3477
  nsenc: 954269209
```
but I want the secs is like 1639390007.73391914 since 1. Jan 1970,
then I followed #74 , I set  the ouster timstamp mode to TIME_FROM_PTP_1588 via
```sh
nc 192.168.1.10 7501
get_config_txt
set_config_param timestamp_mode TIME_FROM_PTP_1588
reinitialize
save_config_params
```
and I get this
```sh
nc 192.168.1.10 7501
get_time_info
{"timestamp": {"time": 3477.954269209, "mode": "TIME_FROM_PTP_1588", "time_options": {"ptp_1588": 3477, "sync_pulse_in": 0, "internal_osc": 3468}}, "sync_pulse_in": {"locked": 0, "polarity": "ACTIVE_HIGH", "diagnostics": {"last_period_nsec": 0, "count": 0, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "OFF", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 0}, "nmea": {"locked": 0, "polarity": "ACTIVE_HIGH", "ignore_valid_char": 0, "baud_rate": "BAUD_9600", "leap_seconds": 0, "diagnostics": {"decoding": {"utc_decoded_count": 0, "date_decoded_count": 0, "not_valid_count": 0, "last_read_message": ""}, "io_checks": {"start_char_count": 0, "char_count": 0, "bit_count": 1, "bit_count_unfiltered": 0}}}}}
```

In annother terminal I use 
```sh
sudo ptpd -i enp89s0 -M
```
to sync, and I launch the ouster ros node and check the timestamp 
```sh
rostopic echo /os_cloud_node/points/header/stamp
```

but I still get 
```sh
stamp:
  secs: 3477
  nsenc: 954269209
```

So what should I do if I wanna use ROS timestamp, instead of using sensor own timestamp.

Best,
Xinliang

TurtleZhong, 2022-02-24:
I also followed the user mannual to set ptp4 but I got this one
```sh
sudo systemctl status ptp4l      
● ptp4l.service - Precision Time Protocol (PTP) service
   Loaded: loaded (/lib/systemd/system/ptp4l.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/ptp4l.service.d
           └─override.conf
   Active: failed (Result: exit-code) since Thu 2022-02-24 20:32:24 CST; 4s ago
     Docs: man:ptp4l
  Process: 19611 ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp/ptp4l.conf (code=exited, status=254)
 Main PID: 19611 (code=exited, status=254)

2月 24 20:32:24 subt-pc systemd[1]: Started Precision Time Protocol (PTP) service.
2月 24 20:32:24 subt-pc ptp4l[19611]: unknown option productDescription at line 87 in enp89s0 section
2月 24 20:32:24 subt-pc ptp4l[19611]: failed to parse configuration file /etc/linuxptp/ptp4l.conf
2月 24 20:32:24 subt-pc systemd[1]: ptp4l.service: Main process exited, code=exited, status=254/n/a
2月 24 20:32:24 subt-pc systemd[1]: ptp4l.service: Failed with result 'exit-code'.

```
Which means failed. Below is my ptp4l.conf
```sh
[global]
#
# Default Data Set
#
twoStepFlag             1
slaveOnly               0
priority1               128
priority2               128
domainNumber            0
#clockClass             248
clockClass              128
clockAccuracy           0xFE
offsetScaledLogVariance 0xFFFF
free_running            0
freq_est_interval       1
dscp_event              0
dscp_general            0
#
# Port Data Set
#
logAnnounceInterval     1
logSyncInterval         0
logMinDelayReqInterval  0
logMinPdelayReqInterval 0
announceReceiptTimeout  3
syncReceiptTimeout      0
delayAsymmetry          0
fault_reset_interval    4
neighborPropDelayThresh 20000000
#
# Run time options
#
assume_two_step         0
logging_level           6
path_trace_enabled      0
follow_up_info          0
hybrid_e2e              0
tx_timestamp_timeout    1
use_syslog              1
verbose                 0
summary_interval        0
kernel_leap             1
check_fup_sync          0
#
# Servo Options
#
pi_proportional_const   0.0
pi_integral_const       0.0
pi_proportional_scale   0.0
pi_proportional_exponent        -0.3
pi_proportional_norm_max        0.7
pi_integral_scale       0.0
pi_integral_exponent    0.4
pi_integral_norm_max    0.3
step_threshold          0.0
first_step_threshold    0.00002
max_frequency           900000000
clock_servo             pi
sanity_freq_limit       200000000
ntpshm_segment          0
#
# Transport options
#
transportSpecific       0x0
ptp_dst_mac             01:1B:19:00:00:00
p2p_dst_mac             01:80:C2:00:00:0E
udp_ttl                 1
udp6_scope              0x0E
uds_address             /var/run/ptp4l
#
# Default interface options
#
network_transport       UDPv4
delay_mechanism         E2E
#time_stamping          hardware
time_stamping           software
tsproc_mode             filter
delay_filter            moving_median
delay_filter_length     10
egressLatency           0
ingressLatency          0
boundary_clock_jbod     1
[enp89s0]
#
# Clock description
#
productDescription      ;;
revisionData            ;;
manufacturerIdentity    00:00:00
userDescription         ;
timeSource              0xA0

```

kairenw, 2022-02-25:
Hi @TurtleZhong,

Thanks for contacting us. I believe you've configured your ptp4 wrong. This may take some back and forth so it'll be easier to work with [our support team](https://ouster.atlassian.net/servicedesk/customer/portal/8). You can link this issue and they'll help you set it up right. Thanks!

Closing this ticket now. If you don't get this addressed, feel free to re-open it.

closed 2022-02-25 by kairenw

learnhard22, 2022-08-20:
Hi @TurtleZhong,

did you already solve this problem and configured your ptp4 right, respectively? I am very interested in, because I also wanna use the ros time to synchronize the timestamp between my Ouster OS0-128 and the IMU that I'am using.

Kind regards,
Lennard

TurtleZhong, 2022-10-10:
> Hi @TurtleZhong,
> 
> did you already solve this problem and configured your ptp4 right, respectively? I am very interested in, because I also wanna use the ros time to synchronize the timestamp between my Ouster OS0-128 and the IMU that I'am using.
> 
> Kind regards, Lennard

Hi, you should check your network like this
```
sudo ethtool -T enp89s0 
Time stamping parameters for enp89s0:
Capabilities:
	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
```
and your Capabilities should have software-transmit like this
```
Time stamping parameters for eno2:
Capabilities:
	software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
```
Then everything will be ok!