ouster-lidar/ouster-sdk #353: ros launch replay errer. calling config service failed

tostman-qwer, 2022-02-25:
I'm running this as my roslaunch command
roslaunch ouster_ros ouster.launch replay:=true metadata:=/home/adas/test_data/json/OS-1-32-122139000936-1024x10.json

this is the error that im getting

[ERROR] [1645757728.430965613]: Calling config service failed
================================================================================
REQUIRED process [os_node-1] has died!
process has died [pid 19990, exit code -6, cmd /home/adas/catkin_ws/ou/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/adas/.ros/log/4c874b26-954b-11ec-b560-7085c22d6d97/os_node-1.log].
log file: /home/adas/.ros/log/4c874b26-954b-11ec-b560-7085c22d6d97/os_node-1*.log
Initiating shutdown!
================================================================================
and log

... logging to /home/adas/.ros/log/4c874b26-954b-11ec-b560-7085c22d6d97/roslaunch-adas-desktop-19974.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://adas-desktop:34579/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 
 * /os_node/lidar_port: 0
 * /os_node/metadata: /home/adas/test_d...
 * /os_node/replay: True
 * /os_node/sensor_hostname: 
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 
 * /os_node/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)

ROS_MASTER_URI=http://localhost:11311

process[os_node-1]: started with pid [19990]
process[os_cloud_node-2]: started with pid [19991]
process[relay_lidar-3]: started with pid [19997]
[ INFO] [1645757728.347556514]: Client version: 0.3.0+13ea8e8-release
[ INFO] [1645757728.348399482]: Running in replay mode
terminate called after throwing an instance of 'std::invalid_argument'
  what():  * Line 2, Column 1
  Syntax error: value, object or array expected.

process[relay_imu-4]: started with pid [20002]
[ERROR] [1645757728.430965613]: Calling config service failed
================================================================================REQUIRED process [os_node-1] has died!
process has died [pid 19990, exit code -6, cmd /home/adas/catkin_ws/ou/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/adas/.ros/log/4c874b26-954b-11ec-b560-7085c22d6d97/os_node-1.log].
log file: /home/adas/.ros/log/4c874b26-954b-11ec-b560-7085c22d6d97/os_node-1*.log
Initiating shutdown!
================================================================================
[relay_imu-4] killing on exit
[relay_lidar-3] killing on exit
[os_cloud_node-2] killing on exit
[os_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kairenw, 2022-02-25:
Interesting! Nothing looks obviously wrong to me right now. Mind submitting your metadata file and bag through [our Support portal](https://ouster.atlassian.net/servicedesk/customer/portal/8)? You can link this ticket and mention specifically that I (Kairen) sent you and am willing to work on it.

tostman-qwer, 2022-02-28:
Hi. Thank you for your answer.
I checked the meta data, and it was empty. 
There must have been a problem moving the file.
Thanks!

kairenw, 2022-02-28:
Looking at the log, the most relevant-seeming error is:
```cfg getaddrinfo(): Name or service not known```

This indicates to me that it was not able to reach your sensor. Can you ping the sensor at that hostname (`os1-122139000936.local`) and show me the log? Thanks!

tostman-qwer, 2022-02-28:
sensor hostname is os-xxxx.
so my command:
 roslaunch ouster_ros ouster.launch sensor_hostname:=os-122139000936.local metadata:=/home/adas/meta.json viz:=true lidar_mode:=1024x10 udp_dest:=192.168.0.102
and log
... logging to /home/adas/.ros/log/da679d34-9860-11ec-b940-7085c22d6d97/roslaunch-adas-desktop-8790.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://adas-desktop:40557/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 1024x10
 * /os_node/lidar_port: 0
 * /os_node/metadata: /home/adas/meta.json
 * /os_node/replay: False
 * /os_node/sensor_hostname: os-122139000936.l...
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 192.168.0.102
 * /os_node/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    img_node (ouster_ros/img_node)
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[os_node-1]: started with pid [8805]
process[os_cloud_node-2]: started with pid [8806]
[ INFO] [1646034537.567577858]: Client version: 0.3.0+13ea8e8-release
[ INFO] [1646034537.568253335]: Waiting for sensor os-122139000936.local to initialize ...
[ INFO] [1646034537.568379072]: Sending UDP data to 192.168.0.102
process[rviz-3]: started with pid [8812]
process[img_node-4]: started with pid [8818]
process[relay_lidar-5]: started with pid [8819]
process[relay_imu-6]: started with pid [8825]
[ INFO] [1646034537.664939035]: rviz version 1.13.23
[ INFO] [1646034537.664979389]: compiled against Qt version 5.9.5
[ INFO] [1646034537.664991269]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1646034537.667531526]: Forcing OpenGl version 0.
[ INFO] [1646034537.680135290]: Sensor configured successfully
[ INFO] [1646034538.136840381]: Stereo is NOT SUPPORTED
[ INFO] [1646034538.136891228]: OpenGL device: NVIDIA TITAN X (Pascal)/PCIe/SSE2
[ INFO] [1646034538.136910819]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1646034555.726630131]: Sensor initialized successfully
[ INFO] [1646034555.727253821]: Wrote metadata to /home/adas/meta.json
[ INFO] [1646034555.729192928]: Using lidar_mode: 1024x10
[ INFO] [1646034555.729274794]: OS-1-32-U2 sn: 122139000936 firmware rev: v2.2.1

but why do visualizations to be any more?
[image]

thanks.

kairenw, 2022-02-28:
You've set the udp_dest wrong. Can you try:
```roslaunch ouster_ros ouster.launch sensor_hostname:=os-122139000936.local metadata:=/home/adas/meta.json viz:=true lidar_mode:=1024x10``` instead? That will set your udp_dest automatically. If it doesn't work, we can try to find it manually.Thanks!

kairenw, 2022-03-07:
Closing ticket as no response. Please let us know if you're still having issues.

closed 2022-03-07 by kairenw