ouster-lidar/ouster-sdk #355: How can I improve the number of point clouds in a scan?

Duke-Rodriguez-Rudolph, 2022-02-27:
Hello, I am using your product and I am missing a problem that I can only get a some point clouds of partial angles from each scan, not all angles. As in your example, I need to wait longer to get a more complete result. So how can I get a complete result in a scan or in a very short time?

kairenw, 2022-02-28:
Hi @Duke-Rodriguez-Rudolph,

Thanks for reaching out. 

How are you using our product? Are you using cpp, ROS, or python, and which example are you looking at? Also, it would help to know which version of the firmware you're on! 

A few guesses -- one of these could apply, or it could be something different.
If you're consistently missing reads from certain parts of each scan, then you could check your azimuth window settings. (Go to your sensor homepage at `http://<SENSOR-HOSTNAME>.local`)
If you're getting what looks like a packet loss effect (dropped parts of scans aren't consistent between scans), then it could be a number of things, including your ethernet cable, the routers you're using, or something else. It may be easier to contact our [support team,](https://ouster.atlassian.net/servicedesk/customer/portal/8) in that case, as it is actually a networking error.

Duke-Rodriguez-Rudolph, 2022-03-03:
Thank you for your reply. I use the code example of CPP, and  I follow the example of "ouster_viz". I logged in to your official website, but I only found Python related tutorials in it. Lack of tutorials, I have a little difficulty getting started with CPP.

kairenw, 2022-03-03:
Yes, we hope that one day we can document the cpp a bit better! In the mean time, do you mind following that link above and contacting our support team? I think that will slow down the communication cycles. That way you can also show us pictures of what you're seeing.

Closing, but feel free to re-open.

closed 2022-03-03 by kairenw