ouster-lidar/ouster-sdk #356: How to can I obtain the XYZ coordinates and reflection intensity of a point in the simplest way

stingliang, 2022-03-03:
I want to quickly and simply obtain the coordinate information and reflection intensity information of point cloud when using sensors to collect data in real time, so as to realize the SLAM algorithm of our products. Maybe I didn't read the document carefully enough, but I still hope to get some guidance.

kairenw, 2022-03-03:
Hi @stingliang, 

I'd encourage you to work through the docs for the [Python SDK](https://static.ouster.dev/sdk-docs/installation.html), if you're comfortable working in Python. Obtaining coordinate, reflectivity, and signal information from a point cloud can be done using `client.Scans`, which is introduced in the [Quick Start](https://static.ouster.dev/sdk-docs/quickstart.html), and further explained in the [example showing how to work with the LidarScan class](https://static.ouster.dev/sdk-docs/examples.html#the-lidarscan-representation).

Hope that helps! 

Closing this issue unless you have a specific question about use. In that case, feel free to re-open.

stingliang, 2022-03-03:
I'm sorry that I didn't clarify my requirements. I can successfully obtain the information of point cloud and reflection intensity through Python SDK. 
But I want to obtain the relevant information through C++ code.
Through [example.cpp](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/src/example.cpp), I didn't find a good way to get the reflection intensity information of each point (the coordinate acquisition is very clear in the routine)

kairenw, 2022-03-04:
Oh, I see!

To get reflectivity and signal in cpp, you can use [`field`](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/lidar_scan.h#L155): 
```
auto reflectivity = scan.field(sensor::REFLECTIVITY)
```

That's assuming you have `scan` as a `LidarScan` like in the example cpp code.

For signal, it's the same but you want `sensor::SIGNAL`. You can find the available fields in [`types.h`](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/types.h#L393). Please note that unless your sensor supports dual returns (Rev 6), you won't be able to get the fields like `SIGNAL2` and `REFLECTIVITY2`.

closed 2022-03-04 by kairenw