ouster-lidar/ouster-sdk #357: The calculation of "cartesian()" is too slow

Duke-Rodriguez-Rudolph, 2022-03-05:
When I use "cartesian()", I find that the calculation speed of this function is too slow. I use OS1-64. In your visualization examples, range image and point clouds are computed and show very quickly. Unfortunately, I don't quite understand the underlying logic of the example. Is there any other way to skip "cartesian()" and get the range image directly?

Duke-Rodriguez-Rudolph, 2022-04-06:
Thank you for your reply! During this period of time, I studied this function. I found that it processes all point cloud data and reshapes the matrix. I think it consumes a lot of time. Therefore, I made some modifications and only calculated the part I need and removed the operation of  reshape. The operation time is much faster, and my problem is solved. By the way, I found that it takes a long time to get information from lidar to my computer (I used multithreading to solve this problem). Then I noticed that the information transmission uses the network cable we usually use to connect to the Internet, so will the data transmission be affected by the quality of this network cable?

kairenw, 2022-04-19:
Data transmisison can definitely be affected by the quality of your network cable. It has to support gigabit.

Glad that you got the cartesian function working as you needed. 

Since this seems resolved, I will now close this issue. Thanks!

closed 2022-04-19 by kairenw