ouster-lidar/ouster-sdk #359: xavier agx - cannot record pcap

Leonel-Romero, 2022-03-11:
We are trying to get the ouster-sdk example to record pcap files on a jetson board.

The installation seems to work. We have tried both in and out of a virtual environment. When we run
 "python3 -m ouster.sdk.examples.client $SENSOR_HOSTNAME record-pcap" python connects to the sensor and creates configuration json but then it times out.

See output below, which to us is not very helpful. We are able to run the examples with both the c and ros. Any help would be much appreciated. We did notice that both the c and ros examples take some time to get started, whereas python times out after 1s. 

example: record-pcap
WARNING: No lidar profile found.
Saving sensor metadata to: OS-1-128_122109000759_1024x10_20220310_185417.json
Writing to: OS-1-128_122109000759_1024x10_20220310_185417.pcap (Ctrl-C to stop early)
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/leromero/.local/lib/python3.6/site-packages/ouster/sdk/examples/client.py", line 305, in <module>
    main()
  File "/home/leromero/.local/lib/python3.6/site-packages/ouster/sdk/examples/client.py", line 301, in main
    example(args.hostname)  # type: ignore
  File "/home/leromero/.local/lib/python3.6/site-packages/ouster/sdk/examples/client.py", line 259, in record_pcap
    n_packets = pcap.record(source_it, f"{fname_base}.pcap")
  File "/home/leromero/.local/lib/python3.6/site-packages/ouster/pcap/pcap.py", line 330, in record
    for packet in packets:
  File "/home/leromero/.local/lib/python3.6/site-packages/more_itertools/more.py", line 3281, in __next__
    item = next(self._iterable)
  File "/home/leromero/.local/lib/python3.6/site-packages/ouster/client/core.py", line 218, in __iter__
    self.flush(full=True)
  File "/home/leromero/.local/lib/python3.6/site-packages/ouster/client/core.py", line 263, in flush
    f"No packets received within {self._timeout}s")
ouster.client.core.ClientTimeout: No packets received within 1.0s

kairenw, 2022-03-11:
Hi @Leonel-Romero,

This error is a symptom of a variety of networking issues, so bear with me as we cross out the easy answers. Have you checked that the ports are configured properly, and the `udp_dest` is set properly? You can do that by using the sensor's homepage (http://<SENSOR-HOSTNAME>.local) or using ``python3 -m ouster.sdk.examples.client $SENSOR_HOSTNAME configure-sensor``. 

It'd also help to know how your sensor is connected to your computer.

Leonel-Romero, 2022-03-11:
Thanks for the reply. We checked the configuration and looked fine.
ran: sudo apt upgrade
After rebooting it works fine now. We can record pcap files.

Thanks again.

kairenw, 2022-03-11:
Great! I'm so glad.
Will close out the ticket now.

closed 2022-03-11 by kairenw

bdytx5, 2022-12-03:
seems like having ouster studio open and conncected to the sensor while running python commands can cause this