ouster-lidar/ouster-sdk #360: pcl ring datatype update

girvenavery2022, 2022-03-12:
## OS 
- ubuntu 20.04
## Ros
 - noetic
## information 
The issue I am creating here, goes right along with the issue I created with the Ros2 driver repo (https://github.com/ros-drivers/ros2_ouster_drivers/issues/95). But essentially, the ring data type is defined as a `uint8_t` and while this doesn't break anything, updating it to a `uint16_t` has the benefits of increasing the portability of both the Ros1 and Ros2 Ouster Drivers and allowing the users to create more reusable code because ignition gazebo and other 3D lidar drivers use a `uint16_t` as their ring datatype.
I already have this fixed completed and ready to be tested when I get to my Ouster on Monday. I can also make PR with these changes so please let me know if i need to add anything!

kairenw, 2022-03-21:
Hi @girvenavery2022,

Thanks for pointing that out! That makes sense. We would welcome the PR. If so, could you make a note in the CHANGELOG? The big heading would be [Unreleased], and under that if you could just write `ouster_ros` and then the change under that!

girvenavery2022, 2022-03-23:
Hello!
Awesome, I updated the CHANGELOG on my fork that has the ring type changes and I will open up a PR for it!