ouster-lidar/ouster-sdk #361: Python SDK metadata fetching

d33dler, 2022-03-20:
`  File "/home/x/x/x/x/sensor/ouster_sensor.py", line 95, in __fetch_meta
    with closing(cl.Sensor(hostname=self.host)) as source:
  File "/home/x/x/x/x/venv/lib/python3.8/site-packages/ouster/client/core.py", line 141, in __init__
    self._fetch_metadata()
  File "/home/x/x/x/x/venv/lib/python3.8/site-packages/ouster/client/core.py", line 155, in _fetch_metadata
    raise ClientError("Failed to collect metadata")
ouster.client.core.ClientError: Failed to collect metadata`

I'm getting this error when working with virtualization (sim) of an OS-1-16-A2 sensor and trying to live-stream according to SDK examples. The simulation works as a service installed on a VirtualBox (Ubuntu Server 20.04) that listens on -  ip: VBox network interface ip,  port: 7501. a PCAP file is played with tcpreplay to the same ip on port 7502 as the destination. I'm able to interact from the host machine through netcat and get (get_sensor_info, etc).

d33dler, 2022-03-21:
Does metadata fetching require something more than get_sensor_info and to run streams?

kairenw, 2022-03-21:
Yeah, unfortuantely the current format of the metadata contains more information than just `get_sensor_info` on the sensor will get you. You can read all the parts of what makes the current metadata [here](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/src/client.cpp#L247).

It'd be useful to know if you can get this [cpp example ](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/src/example.cpp)to run on your simulated sensor. See the main README for how to build/run it. That'll help us pinpoint if we should be looking at the way the python sdk is bound, or below that at something in the cpp sdk.

d33dler, 2022-03-29:
Thank you for your reply; issue resolved

closed 2022-03-29 by d33dler