ouster-lidar/ouster-sdk #365: CMake find_package(ouster_client)

vebjornjr, 2022-04-05:
Hi!

What is the simplest way to use the ouster_client code in a existing CMake-based project?

I build this project with CMake as described in the README and in addition installed it with `cmake --build build --target install`.

It seems that the header files and .so file are correctly installed in user/local/include and usr/local/lib, but that this project doesn't support `find_package(ouster_client REQUIRED)`.

Thank you!

bexcite, 2022-04-08:
Hi **vebjornjr**,

I think this should work:

```
find_package(OusterSDK CONFIG REQUIRED)
```
and then use targets ``OusterSDK::ouster_client`` and ``OusterSDK::ouster_pcap``.

Let us know if it solves the issue. Thanks!

kairenw, 2022-04-19:
Hey @vebjornjr

Thanks for the suggestion. We'd definitely like to; it's just stacked behind some other critical needs currently. Thanks for bringing it up, and we'll re-up the priority internally next time we get a chance.

I've renamed this issue, and we'll keep it open until the need is addressed. Thanks!