ouster-lidar/ouster-sdk #367: Decoding "/os_node/lidar_packets" and "/os_node/imu_packets"

JJuhyeon-Kim, 2022-04-11:
Hi, thank you for your hard work.

I have one simple question about decoding "/os_node/lidar_packets" and "/os_node/imu_packets" in bag file.
I understood the concept of packet through issue number 94 ([image]).
BTW, I acquired bag files using packets as below, and I want to convert it to "os_cloud_node/points". 
[image]

I tried "rosbag record -a -O myfile.bag" command before, but I think this command has a problem.
When I tried SLAM algorithm with decoded bag file, it did not work well.

Is there anyone who can help me?

kairenw, 2022-04-11:
Hi @JJuhyeon-Kim,

It would be useful to understand what you are doing exactly, i.e., how are you converting the packets into point clouds? And what is the problem you are getting when you run `rosbag record -a -O myfile.bag`?

JJuhyeon-Kim, 2022-04-20:
> Hi @JJuhyeon-Kim,
> 
> It would be useful to understand what you are doing exactly, i.e., how are you converting the packets into point clouds? And what is the problem you are getting when you run `rosbag record -a -O myfile.bag`?

I acquired bag file with 'lidar_packets' topic, but what I need is XYZ positions of each point.
According to the information from #94 , there seems to be no XYZ information in data with 'lidar_packet' topic. 
Is there any way I can convert the data from 'lidar_packets' to 'os_cloud_node/points'?

Thank you for your kind reply.

kairenw, 2022-04-20:
Yes, if you run the ros commands according to the [README:](https://github.com/ouster-lidar/ouster_example/blob/master/README.rst)
```roslaunch ouster_ros ouster.launch sensor_hostname:=<sensor hostname> metadata:=<path to metadata json>```

that wlil start publishing /os_cloud_node/points

kairenw, 2022-05-04:
Closing due to no further questions. Feel free to re-open if the question is not resolved!

closed 2022-05-04 by kairenw