ouster-lidar/ouster-sdk #368: Calling config service failed

DanielZ98, 2022-04-12:
When i try to replay a rosbag, an error appears

roslaunch ouster_ros ouster.launch replay:=true metadata:=/media/zhaoyu/WD_BLACK03/data/20220119_141034/metadata.json
... logging to /home/zhaoyu/.ros/log/7cbc4e3a-ba86-11ec-a3c5-d3c819544582/roslaunch-zhaoyu-GP66-Leopard-11UG-58576.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zhaoyu-GP66-Leopard-11UG:43455/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 
 * /os_node/lidar_port: 0
 * /os_node/metadata: /media/zhaoyu/WD_...
 * /os_node/replay: True
 * /os_node/sensor_hostname: 
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)

auto-starting new master
process[master]: started with pid [58584]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7cbc4e3a-ba86-11ec-a3c5-d3c819544582
process[rosout-1]: started with pid [58594]
started core service [/rosout]
process[os_node-2]: started with pid [58601]
process[os_cloud_node-3]: started with pid [58602]
process[relay_lidar-4]: started with pid [58603]
process[relay_imu-5]: started with pid [58604]
[ INFO] [1649784732.812797188]: Client version: 0.3.0.dev2+3e6af59-release
[ INFO] [1649784732.814101497]: Running in replay mode
WARNING: No data_format found.
terminate called after throwing an instance of 'std::invalid_argument'
  what():  n_cols_of_lidar_mode
[ERROR] [1649784732.911260941]: Calling config service failed
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 58601, exit code -6, cmd /home/zhaoyu/ouster/ros_ws/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/zhaoyu/.ros/log/7cbc4e3a-ba86-11ec-a3c5-d3c819544582/os_node-2.log].
log file: /home/zhaoyu/.ros/log/7cbc4e3a-ba86-11ec-a3c5-d3c819544582/os_node-2*.log
Initiating shutdown!
================================================================================
[relay_imu-5] killing on exit
[relay_lidar-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

kairenw, 2022-04-12:
Not sure why it's doing that on replay. Do you mind pasting your metadata here? If you don't want to put it on the web, you can also contact us through [Ouster support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86). In your description, you can reference this ticket and say that you'd like to be put through to talk to me (Kairen)

DanielZ98, 2022-04-13:
> Not sure why it's doing that on replay. Do you mind pasting your metadata here? If you don't want to put it on the web, you can also contact us through [Ouster [...]

there is nothing in metadata

kairenw, 2022-04-13:
> there is nothing in metadata

Ah! That would be why. You want to use the metadata that got written out when you recorded the bag there. Otherwise, it can't do the projection from range to XYZ coordinates.

kairenw, 2022-04-19:
Hi @DanielZ98, 

This should be resolved. Feel free to reopen this if the issues continues.

closed 2022-04-19 by kairenw