ouster-lidar/ouster-sdk #373: REQUIRED process [os_node-2] has died!

havvanurbozomeroglu, 2022-04-19:
Hi, I have OS1 lidar. I got some troubles!

i'm running >> roslaunch ouster_ros ouster.launch sensor_hostname:=os1-99xxxxxxxxxx.local udp_dest:=xx  rviz:=true
... logging to /home/hnb/.ros/log/28f6b21e-c00c-11ec-9c38-54ab3ae0f419/roslaunch-hnb-4485.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hnb:xxxxx/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 
 * /os_node/lidar_port: 0
 * /os_node/metadata: 
 * /os_node/replay: False
 * /os_node/sensor_hostname: os1-991946000512....
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: 178.243.238.248
 * /os_node/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)

auto-starting new master
process[master]: started with pid [4498]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 28f6b21e-c00c-11ec-9c38-54ab3ae0f419
process[rosout-1]: started with pid [4512]
started core service [/rosout]
process[os_node-2]: started with pid [4519]
process[os_cloud_node-3]: started with pid [4520]
[ERROR] [1650391901.121287050]: Must specify path for metadata output
process[relay_lidar-4]: started with pid [4526]
process[relay_imu-5]: started with pid [4528]
[ INFO] [1650391901.133029223]: waitForService: Service [/os_node/os_config] has not been advertised, waiting...
================================================================================REQUIRED process [os_node-2] has died!
process has died [pid 4519, exit code 1, cmd /home/hnb/catkin_ws/devel/lib/ouster_ros/os_node __name:=os_node __log:=/home/hnb/.ros/log/28f6b21e-c00c-11ec-9c38-54ab3ae0f419/os_node-2.log].
log file: /home/hnb/.ros/log/28f6b21e-c00c-11ec-9c38-54ab3ae0f419/os_node-2*.log
Initiating shutdown!
================================================================================
[relay_imu-5] killing on exit
[relay_lidar-4] killing on exit
[os_cloud_node-3] killing on exit
[os_node-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Where am I doing wrong? Very happy if you help.

kairenw, 2022-04-19:
Hi @havvanurbozomeroglu 

The relevant line in the logs is here:
```
[ERROR] [1650391901.121287050]: Must specify path for metadata output
```

Essentially, for your command you should specify something like this: 
```roslaunch ouster_ros ouster.launch sensor_hostname:=os1-99xxxxxxxxxx.local udp_dest:=xx rviz:=true metadata:=$PWD/metadata.json```

Hope that helps!

kairenw, 2022-04-19:
I'll keep this open for the time being -- let us know if you have more problems!

closed 2022-04-19 by havvanurbozomeroglu

havvanurbozomeroglu, 2022-04-19:
thanks @kairenw. the ouster.launch was successful in running the visualization but ı can see anything on rviz.
Very happy if you help
[image]
.

kairenw, 2022-04-19:
Sure, let's pick it up in the other ticket.