ouster-lidar/ouster-sdk #374: ı can see anything on rviz

havvanurbozomeroglu, 2022-04-19:
Hi guys,  the ouster.launch was successful in running the visualization but ı can see anything on rviz.
Very happy if you help
[image]

kairenw, 2022-04-19:
Can you paste the entire log from when you run roslaunch?

havvanurbozomeroglu, 2022-04-19:
roslaunch ouster_ros ouster.launch sensor_hostname:=os1-99xxxxxxxxxxxx.local udp_dest:=xxxx  rviz:=true metadata:=$PWD/metadata.json
... logging to /home/hnb/.ros/log/372ebe1c-c02a-11ec-aaf4-54ab3ae0f419/roslaunch-hnb-13915.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hnb:xxxx/

SUMMARY
========

PARAMETERS
 * /os_cloud_node/tf_prefix: 
 * /os_node/imu_port: 0
 * /os_node/lidar_mode: 
 * /os_node/lidar_port: 0
 * /os_node/metadata: /home/hnb/catkin_...
 * /os_node/replay: False
 * /os_node/sensor_hostname: os1-xxxxx....
 * /os_node/timestamp_mode: 
 * /os_node/udp_dest: xxxxx
 * /os_node/udp_profile_lidar: 
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    os_cloud_node (ouster_ros/os_cloud_node)
    os_node (ouster_ros/os_node)
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)

auto-starting new master
process[master]: started with pid [13942]
ROS_MASTER_URI=http://localhost:xxx

setting /run_id to 372ebe1c-c02a-11ec-aaf4-54ab3ae0f419
process[rosout-1]: started with pid [13956]
started core service [/rosout]
process[os_node-2]: started with pid [13963]
process[os_cloud_node-3]: started with pid [13964]
process[relay_lidar-4]: started with pid [13969]
process[relay_imu-5]: started with pid [13975]
[ INFO] [1650404810.185309188]: Client version: 0.3.0+f375a31-release
[ INFO] [1650404810.187125770]: Starting sensor os1-991946000512.local initialization...
[ INFO] [1650404810.187354156]: Will send UDP data to 178.243.238.248
[ INFO] [1650404810.197142985]: Sensor os1-991946000512.local configured successfully
[ INFO] [1650404821.443865148]: Sensor os1-991946000512.local initialized successfully
[ INFO] [1650404821.444104107]: Wrote metadata to /home/hnb/catkin_ws/src/ouster_example/metadata.json
[ INFO] [1650404821.444590824]: Using lidar_mode: 1024x10
[ INFO] [1650404821.444621758]: OS-1-16-A0 sn: 991946000512 firmware rev: v2.2.1

kairenw, 2022-04-19:
Where are you getting the `udp_dest` value from?

havvanurbozomeroglu, 2022-04-19:
udp_dest is my pc ip address. but i think it must be lidar address, right?

kairenw, 2022-04-19:
It may be easier to just leave it blank and let the computer set it automatically

havvanurbozomeroglu, 2022-04-19:
thanks for your help. i am  basic level about ros and ı can't display laserscan topic. how can i do this?

kairenw, 2022-04-19:
sorry so, try this instead without the udp_dest argument specified:
```roslaunch ouster_ros ouster.launch sensor_hostname:=os1-99xxxxxxxxxxxx.local rviz:=true metadata:=$PWD/metadata.json```

havvanurbozomeroglu, 2022-04-19:
i got handle this problem. thanks. 
now, ı have another problem that can't display laserscan topic. how can i do this?
[image]

kairenw, 2022-04-19:
cool! 
so if you haven't made any changes to the source code, i don't believe there are any LaserScan topics for you to subscribe to. You can get a look at the available ros topics with `rostopic list -v`

```$ rostopic list -v

Published topics:
 * /rosout_agg [rosgraph_msgs/Log] 1 publisher
 * /rosout [rosgraph_msgs/Log] 6 publishers
 * /initialpose [geometry_msgs/PoseWithCovarianceStamped] 1 publisher
 * /move_base_simple/goal [geometry_msgs/PoseStamped] 1 publisher
 * /clicked_point [geometry_msgs/PointStamped] 1 publisher

Subscribed topics:
 * /rosout [rosgraph_msgs/Log] 1 subscriber
 * /os1_node/lidar_packets [unknown type] 1 subscriber
 * /os1_node/imu_packets [unknown type] 1 subscriber
 * /tf [tf2_msgs/TFMessage] 1 subscriber
 * /tf_static [tf2_msgs/TFMessage] 1 subscriber
 * /os_cloud_node/points2 [sensor_msgs/PointCloud2] 1 subscriber
 * /os_cloud_node/points [sensor_msgs/PointCloud2] 1 subscriber
 * /img_node/reflec_image [sensor_msgs/Image] 1 subscriber
```

You can see none of those are a `LaserScan` 

Are you trying to get a look at the reflectivity data? That would be `/img_node/reflect_image/`. Try pressing "Add" and adding by topic

havvanurbozomeroglu, 2022-04-19:
ı will study on LaserScan. ı hope ı can solve this problem. if ı can't, ı want to ask about this. thank you for your guidance.

kairenw, 2022-04-19:
Great! I will close this issue now.

closed 2022-04-19 by kairenw