ouster-lidar/ouster-sdk #381: Intensity images to 3D point clouds? OS1

mehdizayene, 2022-05-11:
Hello Everyone,

I am creating this issue as I cannot find anywhere how can I go back from intensity images to point clouds visualization, aren't intensity images displaying depth into 2D Images? 
Also, how can we interpret the greyscale pixels value? are their representing distances ? for example, how can I know the distance of some object given the intensity of OS1 images, and what is the unit used.

Thank you in advance,
Mehdi

kairenw, 2022-05-11:
Hi @mehdizayene,

The intensity image is not the depth displayed in a 2D image. It's the intensity value, reflecting the strength of the signal returning (rather than a measure of the depth/distance). However, there is a 1:1 correspondence between the intensity image and the range image. So if you have something at the pixel `(u,v)` it the intensity image which you're interested in, then you can find it in the range image at the same pixel ``(u,v)``.

Not sure if you're working with the data in Python, C++, or ROS. For the two former, it's easy as easy as using ``field`` on the ``LidarScan`` object. You can read more about how to use the ``LidarScan`` on [our docs](https://static.ouster.dev/sdk-docs/index.html). In ROS if you're working with the ``/points`` topic, you may have to do a conversion from the 1D ordering back into 2D, in effect reversing :https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/ros.cpp#L135. 

Finally, the intensity -- now known and accessible as the field ``SIGNAL``-- is given in number of photons. The range image is distance given in millimeters. 

Hope that helps!

kairenw, 2022-05-11:
I've closed this ticket, but feel free to reopen.

If you have more specific code issues, you can contact us with the code/the specific thing you're trying to do [at Ouster support](https://ouster.atlassian.net/servicedesk/customer/portal/8).

kairenw, 2022-05-11:
Hi @mehdizayene, are you using `ouster_ros` at all? If you run the img_node (you can turn it on with the current `ouster.launch` by using `viz:=true` when launching), then you should be able to get the range image message for free.

Take a look at what [img_node ](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/img_node.cpp) does, and let us know if you have more questions!

mehdizayene, 2022-05-20:
Hi again,

Thank you so much for the very good insights, everything works perfectly now.
Only one more question remains. Is data on range images 16 or 8 bits ? if it is 16 bits how can we go from range to intensity without losing info as I think intensity images are 8 bits encoded.

mehdizayene, 2022-05-20:
Actually, The reason why I am asking this is that I only recorded instensity images for a whole instead of range too. can I retrieve them back by some operation ?

kairenw, 2022-05-20:
Hi @mehdizayene,

Sadly, there is no way to to retrieve range back from intensity. They are different measurements entirely!

All the publisehed img messages are 16 bits.

mehdizayene, 2022-05-20:
Hi ,

Even intensity ones? Why does the value goes from 0 to 255 for each pixel then ?

kairenw, 2022-05-20:
Can you tell me which specific topic you recorded, and which lidar profile you used when recording it? There could be a variety of things you're referring to so I want to answer the right question.

mehdizayene, 2022-05-20:
I am using Ouster OS1 and recording the img_node/intensity_image topic

kairenw, 2022-05-20:
What version of the code are you running? We haven't published `intensity_image` for a long time. 
Also, it would still be useful to know the lidar profile of your sensor. If you go to the sensor homepage, that would be under "UDP Profile Lidar"
If you don't have any of that, then maybe tell me which FW of the sensor you are running

mehdizayene, 2022-06-14:
Hi again,

It turned out that images containing the depth are range images and not intensity images...
I am now recording range images and extracting them with ROS but the result is a 8byte value per pixel so a maximum  value of 255 which can't be the actual depth. Is there a mapping formula ?
Are there any other topics that would give depth image or help generate them ?
Because I am also recording ouster_ros/os1_cloud_node but how can this help me generating images with the actual depth in each pixel...?
Thank you in advance, I really tried so so hard to get these images and no clear instructions are available...

mehdizayene, 2022-06-14:
maybe the fact of getting 8byte values come from the ros command I'm using to extract? What would be the appropriate command to extract the range images properly ?

mehdizayene, 2022-06-14:
I am using this repo to use ROS with the LiDAR : https://github.com/SteveMacenski/ouster_ros1
This is the command I use to extract data from my range topic :
 rosrun image_view image_saver _filename_format:=renge%04d.png image:=/img_node/range_image

kairenw, 2022-06-14:
Great question, it's not! It's [in 4mm resolution](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/img_node.cpp#L143), so you should shift down if you need it in 1mm resolution.

mehdizayene, 2022-06-14:
If I don't change anything in the code, how can I retrieve the actual depths of each pixel from the generated image? 
Otherwise if not possible what should I change exactly...?

kairenw, 2022-06-14:
You can't if you want to use the `range_image` topic. What I'd recommend you do is to pubish your own message with the full resolution.

mehdizayene, 2022-06-14:
Can we please schedule a call...? I really need help for this.

kairenw, 2022-06-14:
For a pixel at (u,v), your range in mm is the value from `range_image` multiplied by 4. You can achieve this by multiplying or by bit shifting. After you recover the range in mm, you will find that all values greater than 260 mm will be represented the same (represented as 260).

The easiest way to understand how to work with the Ouster SDK is through our [SDK documentation](https://static.ouster.dev/sdk-docs/index.html). We're always happy consider user suggestions for additions which will help users work with the SDK and our sensor.

mehdizayene, 2022-06-14:
I am really sorry for the confusion but I don't understand given a 16byte values how would be the max of multiplying values by 4 equal to 260 ? 
If I understood you well I would end up with a max range representation as 260mm (i.e 0.266 meters)?

kairenw, 2022-06-14:
So, you have 2^16 bits, which gives you 65536 possible values. Then you multiply by 4mm, to get 262144 mm, which is approximately 260 meters.

mehdizayene, 2022-06-14:
Okay, very clear now. But this should not be a problem then as LiDAR max range is 120m anyway right ? So in theory this case should not happen.

kairenw, 2022-06-14:
Yes, 260 meters should be more than the max range of any of our sensors right now, though I'm not up to date on the latest max range for any particular line.

mehdizayene, 2022-06-14:
Very good then, so no issues anymore this means that with img_node, we can retrieve all range values without having to create any other topic and just multyplying the 4mm resolution image if I understood everything well.
Thank you very very much