ouster-lidar/ouster-sdk #390: low latency point cloud

shubham-shahh, 2022-06-08:
Hi, @kairenw 
I came across [this](https://github.com/KumarRobotics/ouster_decoder) package by [KumarRobotics](https://github.com/KumarRobotics). 
It has very low latency (<0.2ms) compared to ouster_example (>3ms).
can we incorporate it into this standard driver package

thanks

Samahu, 2022-06-08:
@shubham-shahh Thanks for pointing the package out, we have made some improvements to the current implementation of ouster_ros that should reduce the latency, these changes are not published yet but will be soon. I will close this ticket once the new improvements have been pushed.

Samahu, 2022-08-24:
@shubham-shahh A new release has been published which changes ROS nodes into nodelets. While this reduces latency it does not result in bigger change as it is claimed by the linked package. I did a small investigation and their decoder generates point clouds at a lower latency but with some caveats. So will maintain this ticket open for now and will get back to it ASAP.

shubham-shahh, 2022-08-24:
> @shubham-shahh A new release has been published which changes ROS nodes into nodelets. While this reduces latency it does not result in bigger change as it is claimed by the linked package. I did a [...]

Hi thank you so much for the update, I'll look at the new release, also, can you please name some caveats to be aware of before using the decoder?

Samahu, 2022-08-24:
Sure, when comparing the point cloud generated by the current ouster ros driver vs what the package in reference generates, the resulting point cloud that they generate is incomplete (missing substantial parts).

shubham-shahh, 2022-08-25:
> Sure, when comparing the point cloud generated by the current ouster ros driver vs what the package in reference generates, the resulting point cloud that they generate is incomplete (missing substantial parts).

Hi, does this have anything to do with, what they have mentioned [here](https://github.com/KumarRobotics/ouster_decoder/blob/main/README.md#:~:text=The%20decoder%20fills%20the%20range%20image%20and%20the%20point%20cloud%20during%20each%20packet%20callback%20instead%20of%20doing%20all%20computations%20at%20the%20end%20of%20a%20sweep.%20Each%20pixel%20in%20the%20range%20image%20(each%20point%20in%20the%20point%20cloud)%20has%20the%20following%20fields%3A)

Samahu, 2022-08-25:
@shubham-shahh I am not sure which part you are referring to in the document. In general I don't know exactly -yet- how their decode strategy would contribute to the missing the points in the point cloud generated by their decoder.

versatran01, 2022-08-25:
Hi, I'm the author of ouster_decoder. 
The decoding strategy is simple, just check if there's any jump in the frame and measurement id and erase anything in between.

Samahu, 2022-08-25:
Hi @versatran01, thanks for commenting. I haven't tried to run your package recently so my observation might be outdated. I will give another try ASAP.

versatran01, 2022-08-25:
It currently doesn't work due to the catkin_package issue mentioned 
https://github.com/ouster-lidar/ouster_example/pull/420#issuecomment-1227322532

I made some quick hack to get it work in https://github.com/KumarRobotics/ouster_decoder/tree/fw2.4
but you just need to comment out those broken catkin_package lines in the ouster_ros package.