ouster-lidar/ouster-sdk #393: CMake: Option to not build/install the executable ouster_client_example

vebjornjr, 2022-06-10:
Hey again.

We build and install this library (ouster_client.a) to a folder in our git repo for full control of versions with regards to deployment and such. In such a case it would be useful with a CMake option to not build the executable since we don't want to commit a executable that we don't use into our repo.

If we delete the binary produced, we get CMake errors from:
```
find_package(OusterSDK 20220608 EXACT CONFIG REQUIRED
    PATHS
        ${PROJECT_SOURCE_DIR}/../../external
)
```

> -- Found OusterSDK: /media/nvidia/ssd/steer/external/lib/cmake/OusterSDK/OusterSDKConfig.cmake
> CMake Error at /media/nvidia/ssd/steer/external/lib/cmake/OusterSDK/OusterSDKTargets.cmake:88 (message):
>   The imported target "OusterSDK::ouster_client_example" references the file
> 
>      "/media/nvidia/ssd/steer/external/bin/ouster_client_example"
> 
>   but this file does not exist.  Possible reasons include:
> 
>   * The file was deleted, renamed, or moved to another location.
> 
>   * An install or uninstall procedure did not complete successfully.
> 
>   * The installation package was faulty and contained
> 
>      "/media/nvidia/ssd/steer/external/lib/cmake/OusterSDK/OusterSDKTargets.cmake"
> 
>   but not all the files it references.

kairenw, 2022-06-13:
Yeah, that's not ideal! Thanks for pointing that out. We can probably push out a fix probably this week or next.

Leaving this open till it's addressed. Thanks!

kairenw, 2022-08-24:
I believe this is addressed by https://github.com/ouster-lidar/ouster_example/pull/420. Closing this out.

closed 2022-08-24 by kairenw