ouster-lidar/ouster-sdk #394: Example uses deprecated functions/members

vebjornjr, 2022-06-13:
I see that the example code for `ouster_client` uses the deprecated `headers` member of `LidarScan` to check if the measurements are valid. What is the recommend way to do this? I guess there exists a new/better way to handle since you deprecated the other way.

And I agree with #344, it would be easier to work with your SDK if you could comment on what to replace the deprecated functions/members with.

kairenw, 2022-06-13:
Yeah, it would probably make it easier if we left some comments to that effect! We'll try to do a pass at some point. I'll leave this ticket open until then.

The new way to access the `status` is using [this function](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/lidar_scan.h#L273). Directly above that you'll see `timestamp()` and `measurement_id()` which are the corresponding new functions for getting those from your `LidarScan`.

kairenw, 2022-06-16:
Yes, that's close! So previously, you had a vector where `begin()` would work nicely. Now `status` returns a `LidarScan::Header`, which is an Eigen Array, so the syntax just needs to change a little: 
```
    auto status = scan.status();
    auto n_invalid =
        std::count_if(status.data(), status.data() + status.size(),
                      [](const uint32_t s) { return !(s & 0x01); });
```

closed 2022-07-01 by vebjornjr