ouster-lidar/ouster-sdk #399: "Unit has experienced a packet drop rate above normal threshold"

juliangaal, 2022-07-13:
## Env

* OS1-128
* Ubuntu 20.04
* firmware v2.1.1

## Network Setup

* Static IP profile in Network Manager: 192.168.100.1, 255.255.255.0, NO GATEWAY
* `dnsmasq` ran successfully `sudo dnsmasq -C /dev/null -kd -F 192.168.100.50,192.168.100.200 -i enxa0cec865cdb1 --bind-dynamic` with

    ```
    dnsmasq: started, version 2.80 cachesize 150
    dnsmasq: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth nettlehash DNSSEC loop-detect inotify dumpfile
    dnsmasq-dhcp: DHCP, IP range 192.168.100.50 -- 192.168.100.200, lease time 1h
    dnsmasq-dhcp: DHCP, sockets bound exclusively to interface enxa0cec865cdb1
    dnsmasq: reading /etc/resolv.conf
    dnsmasq: using nameserver 127.0.0.53#53
    dnsmasq: read /etc/hosts - 7 addresses
    dnsmasq-dhcp: DHCPDISCOVER(enxa0cec865cdb1) bc:0f:a7:00:31:08 
    dnsmasq-dhcp: DHCPOFFER(enxa0cec865cdb1) 192.168.100.123 bc:0f:a7:00:31:08 
    dnsmasq-dhcp: DHCPREQUEST(enxa0cec865cdb1) 192.168.100.123 bc:0f:a7:00:31:08 
    dnsmasq-dhcp: DHCPACK(enxa0cec865cdb1) 192.168.100.123 bc:0f:a7:00:31:08 os-122125000036
    ```
    
## Launch(file)

```xml
itz_recorder.launch 
<launch>
  <node pkg="tf" type="static_transform_publisher" name="cam" args="0.2 0 -0.05 0 0 0 1  /base_link /cam  100"/>
  <node pkg="tf" type="static_transform_publisher" name="ouster" args="0 0 0.05 0 0 0 1  /base_link /os_sensor  100"/>
	
  <include file="$(find ouster_ros)/ouster.launch">
    <arg name="sensor_hostname" value="192.168.100.123"/>
    <arg name="lidar_mode" value="1024x10"/>
    <arg name="metadata" value="$(find ouster_ros)/metadata.json"/>
    <arg name="timestamp_mode" value="TIME_FROM_PTP_1588"/> 
  </include>	
</launch>
```
and ROS output

```xml
[ INFO] [1657712839.172386801]: Client version: 0.4.1+64a6cbe-release
[ INFO] [1657712839.173353892]: Starting sensor 192.168.100.123 initialization...
[ INFO] [1657712839.173569207]: Will use automatic UDP destination
[ INFO] [1657712839.208045249]: Sensor 192.168.100.123 configured successfully
[ INFO] [1657712857.460689532]: Sensor 192.168.100.123 initialized successfully
[ INFO] [1657712857.461164482]: Wrote metadata to /home/kbs/chemnitz_ws/src/chemnitz_meta/ouster_example/ouster_ros/metadata.json
WARNING: No lidar profile found.
[ INFO] [1657712857.463196349]: Using lidar_mode: 1024x10
[ INFO] [1657712857.463211117]: OS-1-128 sn: 122125000036 firmware rev: v2.1.1
```

After this, no data is received. From the web interface, the logs state:

```
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 4,
        "id": "0x01000055",
        "level": "WARNING",
        "msg": "Unit has experienced a packet drop rate above normal threshold. Please check that the network has at least 1000 Mbps connection. Common causes of this notice may be 100 or 10 Mbps network connections. ",
        "msg_verbose": "",
        "realtime": "148058317679"
      },
```

and later

```
    "active": [
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 6,
        "id": "0x01000015",
        "level": "WARNING",
        "msg": "Client machine announced it is not reachable on the provided lidar data port; check that udp_dest and udp_port_lidar configured on the sensor matches client IP and port.",
        "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.100.1:35301",
        "realtime": "649407610783"
      },
      {
        "active": true,
        "category": "UDP_TRANSMISSION",
        "cursor": 7,
        "id": "0x01000018",
        "level": "WARNING",
        "msg": "Client machine announced it is not reachable on the provided not reachable on IMU data port; check that udp_dest and udp_port_imu configured on the sensor matches client IP and port.",
        "msg_verbose": "Failed to send imu UDP data to destination host 192.168.100.1:57363",
        "realtime": "649425047455"
      },
```

Could you help me out, here? Thanks

kairenw, 2022-07-14:
Thanks for this report, super thorough. Although you're using the Ouster SDK ros driver, I'm decently sure this is a networking issue. The automatic UDP destination print out from ROS is just telling the sensor to sense its own destination. As such I'm going to point you towards our sensor customer support here: https://github.com/ouster-lidar/ouster_example/issues/399

Closing this now. If you don't get much comment there (very unlikely), you can come back, let me know, and I'll poke them for you.

closed 2022-07-14 by kairenw

juliangaal, 2022-07-15:
This may be an issue with the Network Manager GUI. Using 

```bash
$ sudo ip addr add 192.168.100.1/24 dev $ETHERNET_DEVICE 
```

in the terminal, which does the same as adding a static ip interface in Network Manager *on paper*, resolved the issue.

kairenw, 2022-07-15:
Awesome, glad it worked! Sorry I put the wrong link there before. I've corrected the link to our sensor customer support.