ouster-lidar/ouster-sdk #401: "Network is unreachable"

its-pru, 2022-07-15:
Hello everyone,

I am trying to connect and use two sensors at once. I had it working at one point but now when trying to connect to the second one through it's IP address on ouster studio I am getting an "Network is unreachable" error and I can't find anyone who had the same error. Any help would be greatly appreciated.

its-pru, 2022-07-15:
Also, if I switch which port the first ouster is connected to I get the same error.

kairenw, 2022-07-15:
Hi @its-pru,

This repo only pertains to the Ouster SDK! For support with Ouster Studio, please contact our [customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8).

Closing issue now.