ouster-lidar/ouster-sdk #404: ROS node TF unconnected tree ussue

enheragu, 2022-07-19:
Hi,

I'm having some issue integrating the ros node publisher with the rest of our system. The issue comes from the file `os_cloud_node.cpp` and the way the transforms are published.

Currently in the node the point cloud message is published refered to the `os_sensor` frame as follows below:

```cpp
if (h != ls.headers.end()) {
    for (int i = 0; i < n_returns; i++) {
           scan_to_cloud(xyz_lut, h->timestamp, ls, cloud, i);
        lidar_pubs[i].publish(ouster_ros::cloud_to_cloud_msg(cloud, h->timestamp, sensor_frame));
    }
}
```

The specific line is the following:
https://github.com/ouster-lidar/ouster_example/blob/b9bc5c3bd459fa1f04098021c764ecafb35e1408/ouster_ros/src/os_cloud_node.cpp#L91

The thing is that in runtime the only frames that are published are `os_lidar` and `os_imu`. Following line and below:
https://github.com/ouster-lidar/ouster_example/blob/b9bc5c3bd459fa1f04098021c764ecafb35e1408/ouster_ros/src/os_cloud_node.cpp#L109

I am not sure whether the message is published on purpose on the `os_sensor` frame or should/could be changed to `os_lidar` instead. If it is published there on purpose, what exact transform should be use to transform those messages to the `os_lidar`?

many thanks! best

Samahu, 2022-07-28:
Hi @enheragu

Ouster-provided drivers and visualizers represent data in the Sensor Coordinate Frame. Thus, when producing the point cloud messages we just set the frame id to **os_sensor** and no extra transformation is needed.

Both **os_lidar** and **os_imu** are child transforms of **os_sensor**, thus all you need to get the point cloud with reference to these two frames is to apply their respective transforms as shown below:

```C++
#include <tf2_sensor_msgs/tf2_sensor_msgs.h>
#include <sensor_msgs/PointCloud2.h>
#include <geometry_msgs/TransformStamped.h>

const geometry_msgs::TransformStamped tf_os_imu;
const geometry_msgs::TransformStamped tf_os_lidar;

void handle_point_cloud(const sensor_msgs::PointCloud2::ConstPtr& cloud_in) {
   sensor_msgs::PointCloud2 cloud_out;
   tf2::doTransform (cloud_in, cloud_out, tf_os_lidar); //for more information tf2::doTransform refer to 
   // http://docs.ros.org/en/kinetic/api/tf2_sensor_msgs/html/namespacetf2.html#aed94c19f6b57d3c3071670d9ad7c1be6 
 }
```

Let us know if more help is needed

closed 2022-07-28 by Samahu

enheragu, 2022-07-29:
Hi @Samahu 
We saw that apparently **os_sensor** frame is actually in the middle of the sensor, where the LIDAR is so it is the correct frame in fact. **os_lidar** frame seems to be in the top of the sensor. We added **os_sensor** to our URDF as expected and now is everything connected. The confusion came from assuming **os_lidar** was the position where the measures were taken, and **os_sensor** some common frame.
Sorry for the disturbance and many thanks for the answer!