ouster-lidar/ouster-sdk #405: Big timing differences using ptpd with ouster

its-pru, 2022-07-22:
Hey Everyone!

I am running into trouble using ptpd with my a-128. I am currently using ptpd and it seems to be working almost well.

I wrote a script that shows me the difference in time between the system and the os_cloud_node/imu/header to see how well ptpd is working. The problem is sometimes the difference between the two can get upwards of 5 milliseconds and I believe at that point messages will be synched incorrectly. I would think with ptp that the difference should be between nanoseconds and microseconds.

Can anyone let me know what there offsets look like and if they are similar to mine, or if someone had a similar issue and was able to solve it. Thanks for any help!

Samahu, 2022-07-28:
@its-pru Are you using a dedicated hardware PTP device as a master?

I am using PTP enabled NIC with the sensor directly linked to the PC, in my experience the time difference - assuming everything in your setup is right - can vary between 5 to 20 milliseconds. The number of network relays and how far the sensor is from the PTP source can be a factor too.

So if you are seeing a 5-millisecond difference I would say that's is not out of norm.

Samahu, 2022-07-29:
Closing this ticket. If this doesn't fully answer your question let us know.

closed 2022-07-29 by Samahu