ouster-lidar/ouster-sdk #407: No packets received within 1.0s

Sergei1993SS, 2022-07-28:
Good afternoon

I have two sensors OS-0-64-U02.
Both have configured UDP Port Lidar and UDP Port IMU (different on each lidar), they are connected via a gigabit switch. I set up a static IP for the lidars so that they are on the same subnet with the computer (Ubuntu 18.04). I started the stream in different streams (Python SDK) and for some time they worked, but one day one lidar stopped working, and the second one works. giving the error "No packets received within 1.0s". At the same time, simple-viz for a non-working lidar works fine.

At the same time, there were other incomprehensible things with the receipt of packets from the lidar, which either appeared or disappeared.
Tell me, what could be the matter?
Thank you!

kairenw, 2022-07-28:
Hi @Sergei1993SS,

This sounds like a sensor support issue. Go ahead and report your issues [here ](https://ouster.atlassian.net/servicedesk/customer/portal/8)-- and they'll probably want a diagnostic bin, available through your sensor homepage.

I'll close this now, but feel free to re-open if this isn't resolved in the future.

closed 2022-07-28 by kairenw

bdytx5, 2022-12-03:
seems like having ouster studio open and conncected to the sensor while running python commands can cause this

kairenw, 2022-12-15:
Yes, unless you multicast, only one application will consume