ouster-lidar/ouster-sdk #412: What is the difference between intensity and reflectivity?

ywfwyht, 2022-08-05:
hi，guys.

What is the difference between intensity and reflectivity? Can they be interconverted?

Samahu, 2022-08-05:
From our guide:

* **Signal Photons [16 bit unsigned int]** - signal intensity photons in
the signal return measure- ment.
* **Reflectivity [16 bit unsigned int]** - sensor Signal Photons
measurements are scaled based on measured range and sensor sensitivity at
that range, providing an indication of target reflectivity.

The reported Reflectivity is a by-product of range and signal intensity
that gives the user an indication of the target reflectivity.

ywfwyht, 2022-08-08:
@Samahu 
Thank you for your reply,  Intensity and reflectivity can convert to each other? How to convert？

Samahu, 2022-08-08:
The relation is shown by the following equation but it is merely an approximation:
$$reflectivity = α * intensity * range^2$$
Where $α$ is a constant

closed 2022-08-08 by Samahu

ywfwyht, 2022-08-09:
OK! Thank you very much.