ouster-lidar/ouster-sdk #413: How to get live data stream by using CPP sdk?

E-Dreamer-LQ, 2022-08-05:
I found that use example.cpp can get  n scans's info, but I want to get runtime stream data by using cpp sdk. Looking forward to a quick reply.

Samahu, 2022-08-05:
We don’t have a standalone example that shows how to stream live data at
the moment but if you are a bit familiar with ROS you can refer to the
‘os_node.cpp’ which does show how to connect to a sensor and stream imu and
LiDAR data from it, the ‘os_cloud.cpp’ code file shows how to convert the
the LiDAR packets to point cloud.

Hope this helps
-- 
*Ussama Naal*
San Francisco, CA
<http://ouster.com/>

Samahu, 2022-08-11:
@E-Dreamer-LQ I haven't heard back from you so I don't know if my response was sufficient. So I will close this issue for now, you may re-open the issue again if you still need help with it.

closed 2022-08-11 by Samahu