ouster-lidar/ouster-sdk #415: Getting a scan with the nth function of more-itertools is very slow

siddas27, 2022-08-09:
Getting a frame with the `nth` function of `more-itertools` is very slow. Is there any better way of getting a single scan from `ouster.client.Scans`? 

It would be better if we could make `ouster.client.Scans` object an iterator.

dgarbuzov, 2022-08-09:
`ouster.client.Scans` is already an iterable. If you're reading data from a running sensor, scans arrive at either 10 or 20hz, so reading the nth scan from a stream can take n * 100ms. If you're reading from a packet capture, seeking to the nth scan will read and process the entire file up to that point, which might be several gb of data. If you need fast random access into a collection of scans, you probably want to read them into memory first.

siddas27, 2022-08-09:
I installed the latest version of ouster-sdk (0.4.1) available in [pypi](https://pypi.org/project/ouster-sdk/#history). And I am getting the following error if I do next(ouster.client.Scans(pcap_file)) -

```   scan = next(scans)
TypeError: 'Scans' object is not an iterator
```

dgarbuzov, 2022-08-09:
It's iterable but not an iterator, so you need to call `iter()` to get an iterator or use a `for` loop to read scans. Arguably `Scans` should be an iterator as well.

siddas27, 2022-08-09:
I see. Thanks.

closed 2022-08-09 by siddas27