ouster-lidar/ouster-sdk #416: Problems with multiple sensors in one network

themightyoarfish, 2022-08-09:
We have several OS1 sensors in one network. Obviously, each device needs its own separate lidar and imu port, so can't use the defaults.
I have made the observation that if I configure one device with ports X/Y and retrieve some data, then quit my program and configure another device with the same ports X/Y, the data doesnt arrive at all, or only very rarely. 

My first thought was that the first device is still streaming UDP packets to the host on the ports now configured to be also used by the second device and thus they collide and the client cannot assemble them correctly.

Is this a correct assessment? Can anything be done short of requiring distinct ports for each serial number even if only one device is used at a time? Can I tell the sensor to stop sending via the API so I at least liberate the ports for clean shutdowns?

themightyoarfish, 2022-08-09:
I am trying to check if UDP data arrives with `sudo tcpdump -i <iface> -n udp port <port> -X`, but while i receive nothing on the lidar port, i do receive packets on lidar port +1 which is what i use for IMU.

Samahu, 2022-08-09:
@themightyoarfish Yes your assumption that the first device is still sending data on the same ports.

There is probably multiple ways you can resolve this but to me it seems that you can put the first device in **STANDYBY** mode. Thus it would stop streaming.

Samahu, 2022-08-09:
Another option is to use `init_client(sensor_hostname, lidar_port, imu_port)` again on the first sensor before shutdown and pass port numbers that are currently free and won't be used by the second device.

themightyoarfish, 2022-08-10:
Thanks for confirming. Might be worth adding to documentation, although I'm not sure where this would need to go.

closed 2022-08-10 by themightyoarfish

themightyoarfish, 2022-08-10:
Is it possible to set the operating mode to standby without waiting for the complete reconfigration? would be very annoying having to wait for several sensors to reconfigure during application shutdown.

Samahu, 2022-08-10:
Unfortunately, I don't think it is possible yet to do so. However, you could try using the `sensor::get_config/set_config` methods to change the ports and see if this works better for you. Refer to the `config_example.cpp` for usage details.

themightyoarfish, 2022-12-09:
I've had another issue related to this, although it's probably caused more by hardware/network issues. In this case, we have VPN connectivity to the network the sensor is in, and somehow a client configured it to send the UDP data over the VPN. Even when the client was long disconnected, the router would forward the packets to the internet, completely saturating our bandwidth.

So yeah, a functionality to just tell the sensor to shut it without incurring costly reinitialization would've helped here also.

Samahu, 2022-12-09:
@themightyoarfish thanks for the feedback, I did make our product team aware of this request.