ouster-lidar/ouster-sdk #417: save out near_ir images for labelling

bensail-ai, 2022-08-11:
I would like to save out near_ir images from the scans for computer vision labelling. However the near_ir sensor is not equalised to 0-255 standard for images. I have tried converting to uint8 and 16 with no success. Example outputs are attached. 

example of my code below:
scans=client.Scans(source)
scan = nth(scans,84)
IR=scan.field(client.ChanField.NEAR_IR)
IR_img = IR_destaggered.astype(np.uint16)
cv2.imwrite('test2',IR_img)
[image]

bensail-ai, 2022-08-12:
I thought that might be the case. However I tested with the sample data. In the Ouster-viz programme the near_ir images look great. However when I try to replicate with my own script I get the above image. How is it done in ouster-viz?

kairenw, 2022-08-12:
Sorry -- I looked at the image and while I read the word "equalized" I was thinking it was "overexposed" (as you can see from the answer I gave about brightness outside/the sun).

`NEAR_IR` is _also_ not equalized on the sensor, but you can apply a bit of image processing after the fact, and we in fact apply some to make `NEAR_IR` appear better. These are subject to change, so please don't depend on them permanently but we have in `ouster._utils` the `AutoExposure` and `BeamUniformityCorrector`. These are bindings to the following C++ algorithms: https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/image_processing.h

You an see how we're applying them in the code for the ouster-viz: https://github.com/ouster-lidar/ouster_example/blob/master/python/src/ouster/sdk/viz.py#L79

kairenw, 2022-08-12:
The `AutoExposure` does help with overexposed as well, unless the image has just lost too much information. What I'm seeing is an almost all-white image, so I'm not sure it will help with the areas that look like that, but in other places it will scale the contrast so that the bulk of it ends up between the specified percentiles. Useful if you want to do labeling or any CV work on it for sure.

kairenw, 2022-08-24:
Closing this out now. Feel free to re-open with questions.

closed 2022-08-24 by kairenw