ouster-lidar/ouster-sdk #418: About timestamp_mode

vislero, 2022-08-19:
Hello

We tried to use ouster for mav indoor(can't use GPS) navigation in Jetson Xavier NX(jetpack 4.6.1)platform, and we have some questions about timestamp mode.

1. PTP mode
> First of all, we set the timestamp mode to TIME_FROM_PTP_1588, and we tested time sync with below commands(eth0(ouster) : link-local mode)

```
sudo phc2sys -m -c eth0 -s CLOCK_REALTIME -O 0 -u 1 -w
sudo ptp4l -m -H -i eth0 -f /etc/linuxptp/ptp4l.conf
```
> However, sometimes clock drift aware like below image
>[image]
> If you look at the picture, the RMS value suddenly increases, why does the drift occur?

2. OSC mode

> Secondly, we set the timestamp mode to TIME_FROM_INTERNAL_OSC 
> and we modifed your's ros.cpp code like below, for acquiring timestamp synchronized with roscore
>[image]

> But when we check time offset using below command, we noticed a strange test result.
```
rostopic echo --offset /os_cloud_node/points/header
```
>When we recorded the offset and figured it, we found that the offset was steadily increasing and occasionally bounced. like below figure.
>[image]

>  How to get great timestamp for navigation using OSC mode ? 

Best regards

Samahu, 2022-08-19:
Hi @vislero, I will be happy to help.

1. With regard to PTP, if the clock is drifting over time then this indicates a problem in the services configuration. You could reach to our support team and they could iterate with you on what might be wrong.
  - A common mistake is that when configuring ptp4l the values in the `ptp4l.conf`:
  ```
  boundary_clock_jbod 1
  [eno1]
  [eno2]
  ```
  Is that these values are not added to the end of the file as the guide indicates. So please double check 
  that.
  - Another thing is try to restart all affected services again:
  ```bash
  sudo systemctl daemon-reload
  sudo systemctl restart ptp4l
  sudo systemctl restart phc2sys
  ```
  and check for any status errors.
  ```bash
  sudo systemctl status ptp4l
  sudo systemctl status phc2sys
  ```

2. With regard to the drift that you are observing with ROS, I have few questions:
  - Are using a release build?
  - How long the time is for the plot you attached with regard to the topic `/os_cloud_node/points/`
I am asking these questions, because for non release builds the generation of point cloud take quite a bit causing an increase in the message queue of `lidar_packets`. The being said the delay should stabilize after couple of minutes but the main problem remains which is you have a backlog of messages to be processed. This could also happen if you have a machine with insufficient processing capacity or if you are running other processes that are constraining your CPU. So first double check that you are using a release build, then check when running the code again if you are CPU/Memory bound.

Best

Samahu, 2022-08-19:
A side note, we do have an upcoming release (very soon) that gives the option to use ros::time as the timestamp for generated point cloud and imu messages. So please keep an eye on that.

learnhard22, 2022-08-21:
> A side note, we do have an upcoming release (very soon) that gives the option to use ros::time as the timestamp for generated point cloud and imu messages. So please keep an eye on that.

Hey @Samahu,

do you have a precise time frame when the new release is coming?
I'm heavily looking forward to use ros time as timestamp.

Best regards,
Lennard

vislero, 2022-08-22:
Hello @Samahu,

First of all thank you for kind reply.

1. PTP
- With regard to configuration of ptp4l.conf, below is context of my  ptp4l.conf file. could you confirm the mistake in the file ? 
```
[global]
#
# Default Data Set
#
twoStepFlag		1
slaveOnly		0
priority1		128
priority2		128
domainNumber		0
#clockClass		248
clockClass		128
clockAccuracy		0xFE
offsetScaledLogVariance	0xFFFF
free_running		0
freq_est_interval	1
dscp_event		0
dscp_general		0
#
# Port Data Set
#
logAnnounceInterval	1
logSyncInterval		0
logMinDelayReqInterval	0
logMinPdelayReqInterval	0
announceReceiptTimeout	3
syncReceiptTimeout	0
delayAsymmetry		0
fault_reset_interval	4
neighborPropDelayThresh	20000000
#
# Run time options
#
assume_two_step		0
logging_level		6
path_trace_enabled	0
follow_up_info		0
hybrid_e2e		0
tx_timestamp_timeout	1
use_syslog		1
verbose			0
summary_interval	0
kernel_leap		1
check_fup_sync		0
#
# Servo Options
#
pi_proportional_const	0.0
pi_integral_const	0.0
pi_proportional_scale	0.0
pi_proportional_exponent	-0.3
pi_proportional_norm_max	0.7
pi_integral_scale	0.0
pi_integral_exponent	0.4
pi_integral_norm_max	0.3
step_threshold		0.0
first_step_threshold	0.00002
max_frequency		900000000
clock_servo		pi
sanity_freq_limit	200000000
ntpshm_segment		0
#
# Transport options
#
transportSpecific	0x0
ptp_dst_mac		01:1B:19:00:00:00
p2p_dst_mac		01:80:C2:00:00:0E
udp_ttl			1
udp6_scope		0x0E
uds_address		/var/run/ptp4l
#
# Default interface options
#
network_transport	UDPv4
delay_mechanism		E2E
time_stamping		hardware
tsproc_mode		filter
delay_filter		moving_median
delay_filter_length	10
egressLatency		0
ingressLatency		0
boundary_clock_jbod	1
#
# Clock description
#
productDescription	;;
revisionData		;;
manufacturerIdentity	00:00:00
userDescription		;
timeSource		0xA0`
```

- Regards to error check and restart, we didn't set phc2sys, ptp4l as startup program(using daemon), so we can always check the error occur but not,  and we reboot system and restart the same processing many time but RMS drift always aware ...

2. ROS time offset 
- We used master version of ouster_example and we attached results of ```$git log ``` command. 
[image]

- We recorded  using ```$rostopic echo --offset /os_cloud_node/points/header```  for 38 minute with '512x20' lidar_mode. 

- During test, we monitored system resources with the 'jtop' and 'htop' commands. There was no shortage of CPU or memory.

If the next release version you commented is available asap, it would be appreciated.

Thank you.

Samahu, 2022-08-22:
> Hey @Samahu,
>
> do you have a precise time frame when the new release is coming? I'm heavily looking forward to use ros time as timestamp.
>
> Best regards, Lennard

I'd would expect the release to go out this week. I will keep this ticket open till the release is out.

Samahu, 2022-08-22:
@vislero Here are my responses to the two issues you are dealing with:

* I looked briefly at your `ptp4l.conf` and noted that you haven't actually definined the `boundary_clock_jbod` at the end of the fie along with your network interface as described in this step in the [PTP Quickstart Guide](https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/ptp-quickstart.html#configuring-ptp4l-for-multiple-ports).

* With regard to your ROS time delay/drift, I think you missed my point when I asked if you are using a release build. I was referring to release vs debug build. When you build the ouster_ros driver did you specify that the build type as `Release`. If you don't do so it will default to `Debug` build which will run much slower. Please make sure that you invoke **catkin_make** as follows:
  ```bash
  catkin_make --cmake-args -DCMAKE_BUILD_TYPE=Release
  ```
  and check if this helps with the delay.

Finally, as I told Lennard with regard to our own implementation of messages based on ros time, the release should be out this week.

vislero, 2022-08-23:
@Samahu 
- If you see default interface tab in my ptp4l.conf file, ```boundary_clock_jbod```  is  obiously defined as 1. You mean that parameter should move to the end of file or  need same line at the end of file? 
-  Whit reagrad to release build, sorry for the misunderstanding. But we already did  ```Release``` build with ```catkin build -DCMAKE_BUILD_TYPE=Release ``` command. So I think build option is not a consideration.

Thank you :)

Samahu, 2022-08-23:
> @Samahu
>
> * If you see default interface tab in my ptp4l.conf file, `boundary_clock_jbod`  is  obiously defined as 1. You mean that parameter should move to the end of file or  need same line at the [...]

* Yes the `boundary_clock_jbod` parameter and the network interfaces has to be defined at the end of the file.

* With regard to ROS, I guess you could try the following to help bring delay down:
  - Choose a lower lidar_mode setting such as 512x10, or 512x20
  - Utilize ROS Nodelets which should reduce the latency between dependent nodes
Now with regard to the second option (nodelets), if you could hold on for a day or two, then this is  included in the pending release along with other improvements.

vislero, 2022-08-24:
@Samahu 

- With regards to ```boundary_clock_jbod``` ,  I did what you said and got reasonable timestamps in PTP sync mode.
- I found new version of ouster_example that have many update you mentioned. 

Thank you !

Samahu, 2022-08-24:
Great, glad to help.

closed 2022-08-24 by Samahu