ouster-lidar/ouster-sdk #419: Change the format into velodyne format

yulan0215, 2022-08-21:
Hi everyone:

I am sorry that I am fresh and I would like to ask that how can I change the format of ouster into Velodyne like format. I found the code here:
<html>
<body>

void ouster_driver::OS1::convert2XYZIR | ( | const CloudOS1 & | in,
  |   | CloudOS1XYZIR & | out
  | )

but I did not know which code and which part of the code should I change, thx!

<br class="Apple-interchange-newline">
</body>
</html>void ouster_driver::OS1::convert2XYZIR	(	const [CloudOS1](http://docs.ros.org/en/melodic/api/ouster_driver/html/namespaceouster__driver_1_1OS1.html#a9455ea14ce69167cfdc9a34150a1fa31) & 	in,
[CloudOS1XYZIR](http://docs.ros.org/en/melodic/api/ouster_driver/html/namespaceouster__driver_1_1OS1.html#ac346ac2772c984d044b5d6c3d808b3fa) & 	out 
)

kairenw, 2022-08-27:
Hi @yulan0215,

I don't believe you're working with code that Ouster has put up as part of this repo before. I think it may be this repo instead: https://github.com/CPFL/ouster, which is code which was forked from this repo, but which then added the various `convert` functions that you are interested in. You can find them at that repo at this link: https://github.com/CPFL/ouster/blob/autoware_branch/src/os1_ros.cpp. 

Hope that helps! 

Closing this issue as it does not concern this repo's code.

yulan0215, 2022-08-28:
> Hi @yulan0215,
>
> I don't believe you're working with code that Ouster has put up as part of this repo before. I think it may be this repo instead: https://github.com/CPFL/ouster, which is code which [...]

Thx for your quick reply!

But when I configured the *.launch file and run the driver I found the problem:
[image]

Part of my launch file is like this:
<launch>

  <arg name="lidar_address" default="192.168.2.87" doc="hostname or IP address in dotted decimal form of the Ouster sensor"/>
  <arg name="pc_address" default="192.168.2.17" doc="hostname or IP address of the computer (PC) where the sensor will send data packets"/>
  <arg name="lidar_port" default="7501" doc="port to which the sensor should send lidar data"/>
  <arg name="imu_port" default="0" doc="port to which the sensor should send imu data"/>
  <arg name="replay" default="false" doc="when true, the node will listen on ~/lidar_packets and ~/imu_packets for data instead of attempting to connect to a sensor"/>
  <arg name="scan_dur_ns" default="100000000" doc="nanoseconds to batch lidar packets before publishing a cloud"/> 
  <arg name="pointcloud_mode" default="XYZIR" doc="Point cloud mode, supported modes: NATIVE, XYZ, XYZI, XYZIR, XYZIF, XYZIRF, XYZIFN, XYZIRFN (I=intensity, R=ring (in ascending order), F=reflectivity, N=ambient noise). NATIVE is XYZIRFt where t=point timestamp and ring follows Ouster's original descending order."/>

I did not know what happene to this ros driver? Can you do me a favor of it? Thx!

yulan0215, 2022-08-28:
Part of my os1.launch file:

https://github.com/ouster-lidar/ouster_example/issues/419#issuecomment-1229427069

kairenw, 2022-08-28:
Hi Yulan, do you know which code you installed? Ours located here: https://github.com/ouster-lidar/ouster_example or the code here: https://github.com/CPFL/ouster ? 

Also, what version of the FW are you on?

yulan0215, 2022-08-28:
> Hi Yulan, do you know which code you installed? Ours located here: https://github.com/ouster-lidar/ouster_example or the code here: https://github.com/CPFL/ouster ?
> 
> Also, what version of the FW are you on?
Thanks for your quick reply!
I used the FW 2. So the https://github.com/CPFL/ouster  I could not use...

I will appreciate if you can give me some ideas,thx

kairenw, 2022-08-29:
Ah, it would actually be helpful if you gave me the full firmware, i.e., "FW 2.3" or "FW 2.2". If you go to to your sensor's homepage, it should tell you.

What I would suggest is that you clone our repo, and copy the function here: https://github.com/CPFL/ouster/blob/autoware_branch/src/os1_ros.cpp#L240 into https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/ros.cpp. At that point, you can slide it into `cloud_to_cloud_msg` much in the way they do in the other repo [here](https://github.com/CPFL/ouster/blob/autoware_branch/src/os1_ros.cpp#L70). 

Your published cloud will then be in the desired format.

dgarbuzov, 2022-08-30:
Might not be a bad idea to figure out how to support configurable point types on the the cloud topics. Few things I can think of:
- no signal field for the low-bandwidth profile
- sparse vs. structured clouds for more efficient dual returns support
- as suggested here,  float x, y, z, intensity; uint16_t ring; float time for compatibility with existing code with other devices

Likely going to need something to support more output profiles in the future :thinking: @kairenw @Samahu

yulan0215, 2022-09-01:
Thx and actually I used previoud version of ouster lidar driver and the os1_ros.cpp was:
#include <pcl_conversions/pcl_conversions.h>
#include <ros/ros.h>
#include <tf2/LinearMath/Transform.h>
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
#include <cassert>

#include "ouster/os1.h"
#include "ouster/os1_packet.h"
#include "ouster/os1_util.h"
#include "ouster_ros/os1_ros.h"

namespace ouster_ros {
namespace OS1 {

using namespace ouster::OS1;

bool read_imu_packet(const client& cli, PacketMsg& m,
                     const OS1::packet_format& pf) {
    m.buf.resize(pf.imu_packet_size + 1);
    return read_imu_packet(cli, m.buf.data(), pf);
}

bool read_lidar_packet(const client& cli, PacketMsg& m,
                       const OS1::packet_format& pf) {
    m.buf.resize(pf.lidar_packet_size + 1);
    return read_lidar_packet(cli, m.buf.data(), pf);
}

sensor_msgs::Imu packet_to_imu_msg(const PacketMsg& p, const std::string& frame,
                                   const ouster::OS1::packet_format& pf) {
    const double standard_g = 9.80665;
    sensor_msgs::Imu m;
    const uint8_t* buf = p.buf.data();

    m.header.stamp.fromNSec(pf.imu_gyro_ts(buf));
    m.header.frame_id = frame;

    m.orientation.x = 0;
    m.orientation.y = 0;
    m.orientation.z = 0;
    m.orientation.w = 0;

    m.linear_acceleration.x = pf.imu_la_x(buf) * standard_g;
    m.linear_acceleration.y = pf.imu_la_y(buf) * standard_g;
    m.linear_acceleration.z = pf.imu_la_z(buf) * standard_g;

    m.angular_velocity.x = pf.imu_av_x(buf) * M_PI / 180.0;
    m.angular_velocity.y = pf.imu_av_y(buf) * M_PI / 180.0;
    m.angular_velocity.z = pf.imu_av_z(buf) * M_PI / 180.0;

    for (int i = 0; i < 9; i++) {
        m.orientation_covariance[i] = -1;
        m.angular_velocity_covariance[i] = 0;
        m.linear_acceleration_covariance[i] = 0;
    }
    for (int i = 0; i < 9; i += 4) {
        m.linear_acceleration_covariance[i] = 0.01;
        m.angular_velocity_covariance[i] = 6e-4;
    }

    return m;
}

void convert2XYZIR(const CloudOS1& in, CloudOS1XYZIR& out) 
{
   out.points.clear();
   PointXYZIR q;
   for (auto p : in.points) {
       q.x = p.x;
       q.y = p.y;
       q.z = p.z;
       q.intensity = ((float)p.intensity/65535.0)*255.0; //velodyne uses values in [0..255] range
       q.ring = pixels_per_column - p.ring; //reverse the ring order to match Velodyne's (except NATIVE mode which respects Ouster original ring order)
       out.points.push_back(q);
   }
}

sensor_msgs::PointCloud2 cloud_to_cloud_msg(const CloudOS1& cloud, ns timestamp,
                                            const std::string& frame) {
    sensor_msgs::PointCloud2 msg{};
    //pcl::toROSMsg(cloud, msg);
    CloudOS1XYZIR cloud_aux;
    convert2XYZIR(cloud, cloud_aux);
    pcl::toROSMsg(cloud_aux, msg);
    msg.header.frame_id = frame;
    msg.header.stamp.fromNSec(timestamp.count());
    //msg.header.stamp = ros::Time::now();
    return msg;
}

geometry_msgs::TransformStamped transform_to_tf_msg(
    const std::vector<double>& mat, const std::string& frame,
    const std::string& child_frame) {
    assert(mat.size() == 16);

    tf2::Transform tf{};

    tf.setOrigin({mat[3] / 1e3, mat[7] / 1e3, mat[11] / 1e3});
    tf.setBasis({mat[0], mat[1], mat[2], mat[4], mat[5], mat[6], mat[8], mat[9],
                 mat[10]});

    geometry_msgs::TransformStamped msg{};
    msg.header.stamp = ros::Time::now();
    msg.header.frame_id = frame;
    msg.child_frame_id = child_frame;
    msg.transform = tf2::toMsg(tf);

    return msg;
}

I met some problem of pcl library when I was compiling so I commented the /usr/include/pcl-1.8/pcl/conversions.h in line 267:
for_each_type<typename traits::fieldList<PointT>::type> (detail::FieldAdder<PointT>(msg.fields));

I compiled it but then when I recorded the data, the data seems to be strange...

I could not convert bag file to pcl file, the error was: Got 65536 data points in frame /os1_lidar on topic /points_raw with the following fields: Segmentation fault (core dumped)
Can you help me to solve this problem? Thx!

kairenw, 2022-09-01:
When you compiled and recorded the data, did you run the viz and see how it looked?

yulan0215, 2022-09-01:
> When you compiled and recorded the data, did you run the viz and see how it looked?

Thanks for your reply!
Unfortunately, rviz could not show point cloud...

kairenw, 2022-09-01:
What was the error?

yulan0215, 2022-09-01:
> What was the error?

When I ran the driver, no error showed.
In the rviz it showed: Transform [sender=unknown_publisher]
For frame [/os1_lidar]: No transform to fixed frame [world]. TF error: [Could not find a connection between 'world' and 'os1_lidar' because they are not part of the same tree.Tf has two or more unconnected trees.]

kairenw, 2022-09-01:
how are you launching rviz? you can load the rviz config in `config/viz.rviz` Or use `viz:=true` when launching

yulan0215, 2022-09-01:
> how are you launching rviz? you can load the rviz config in `config/viz.rviz` Or use `viz:=true` when launching
I used the command:rviz directly...

kairenw, 2022-09-01:
Great, so go ahead and try repeating your roslaunch command but with `viz:=true` on it

yulan0215, 2022-09-01:
I ran the command: rviz viz:=true but I still get the error:
Transform [sender=unknown_publisher]
For frame [/os1_lidar]: No transform to fixed frame [world]. TF error: [Could not find a connection between 'world' and 'os1_lidar' because they are not part of the same tree.Tf has two or more unconnected trees.]

kairenw, 2022-09-01:
Sorry, I meant _with_ the `ouster.launch` file if you're using that, i.e.: 
`roslaunch ouster ouster.launch sensor_hostname:=XXXXX viz:=true` 
if you're using an older version of the code (which I believe you are)

yulan0215, 2022-09-01:
Thx for your quick reply! 
I am sorry I used this command: roslaunch os1.launch viz:=true (sensor_hostname is in the *.launch file)
But I just successfully start the sensor, RVIZ did not start...

yulan0215, 2022-09-01:
> Sorry, I meant _with_ the `ouster.launch` file if you're using that, i.e.: `roslaunch ouster ouster.launch sensor_hostname:=XXXXX viz:=true` if you're using an older version of the code (which I believe you are)

Thx for your quick reply!
I am sorry I used this command: roslaunch os1.launch viz:=true (sensor_hostname is in the *.launch file)
But I just successfully start the sensor, RVIZ did not start...

kairenw, 2022-09-01:
Can you tell me which version of the code you're running?

kairenw, 2022-09-01:
where did you clone it from and when?

yulan0215, 2022-09-01:
[image]

I met another problem, you can see the picture above, I can get the point cloud but I could not see the point cloud in the rviz...

yulan0215, 2022-09-01:
> [image]
> 
> I met another problem, you can see the picture above, I can get the point cloud but I could not see the point cloud in the rviz...

Seems to be that the format of point cloud had some problem...

kairenw, 2022-09-01:
So, I don't know where the `/points_raw` topic comes from. It's not anything we have in our code, I'm pretty sure

yulan0215, 2022-09-02:
> So, I don't know where the `/points_raw` topic comes from. It's not anything we have in our code, I'm pretty sure

Yes, I used the package modifidied and used for Autoware software so the topic was changed into "/points_raw"

yulan0215, 2022-09-02:
> So, I don't know where the `/points_raw` topic comes from. It's not anything we have in our code, I'm pretty sure

When I did not edit conversions.h in pcl, the error will be shown below:
[image]
Seems to be that pcl library did not work with my modified version....

Samahu, 2022-09-02:
> [image]
> 
> I met another problem, you can see the picture above, I can get the point cloud but I could not see the point cloud in the rviz...

Hi @yulan0215 

If the point cloud is not showing up it does not necessarily means there is a problem with the point cloud itself. To isolate the issue first lets verify that you are getting some data on the topic you referenced `/points_raw`, you can do so by using `rostopic hz  /points_raw` and observe if there are messages being published at all. If not then you can stop here and try to find out why the `/points_raw` is not publishing. next, verify that the messages published on the topic `/points_raw` have valid cloud data, not just empty messages. you can do so using by checking the output of `rostopic echo /points_raw`. If all seems okay then I would suspect that your rviz configuration has some flaws that prevents it from displaying the point cloud properly. One thing that could prevent the point cloud from showing even when valid data are streaming is the reference Frame used. From the picture you posted I see that you are using `os1_lidar`, are using the same value for Current View target frame?

yulan0215, 2022-09-02:
> Hi @yulan0215
>
> If the point cloud is not showing up it does not necessarily means there is a problem with the point cloud itself. To isolate the issue first lets verify that you are getting some [...]

Thx for your quick reply! I will check it later...
However, honestly speaking, the problem is that I need to show the point cloud in the calibration tool: https://github.com/ankitdhall/lidar_camera_calibration
So I am afraid there will be some problem if we have view problem, and by the way, I also modified the conversion. h in PCL. I did not know what will happen, but I will check... Thx!

yulan0215, 2022-09-03:
> Hi @yulan0215
>
> If the point cloud is not showing up it does not necessarily means there is a problem with the point cloud itself. To isolate the issue first lets verify that you are getting some [...]

Hello and thx for your support, unfortunately, the problem even worse when I used the calibration tool to use the point cloud.

The error:
Failed to find match for field 'x'.
Failed to find match for field 'y'.
Failed to find match for field 'z'.
Failed to find match for field 'intensity'.
Failed to find match for field 'ring'.
 Seems to be that the format of the point cloud had problem...

Samahu, 2022-09-07:
> Hello and thx for your support, unfortunately, the problem even worse when I used the calibration tool to use the point cloud.
>
> The error: Failed to find match for field 'x'. Failed to find match [...]

@yulan0215 thanks for reporting the issue, I think I identified the source of the problem that you have observed which I ran into myself (although it manifested differently for me as I was unable to use certain PCL funcitons against our defined **ouster_ros::Point** object). If you want to fix this just move `PCL_NO_PRECOMPILE` pre-processor definition from `ouster_ros/point.h` to `ouster_ros/ros.h` before any pcl includes.  We will work on issuing a hotfix ASAP @kairenw @dgarbuzov

Samahu, 2022-09-15:
@yulan0215 The `unable to find match for field '?'` should be resolved with the last update #432.

Frndznz, 2022-09-21:
Hi, could you add the the conversion to Velodyne style format to the source code? I think it would save many of us unwanted fiddling with this repo and many headaches as well