ouster-lidar/ouster-sdk #426: there is no show on conda environment

oneway3124, 2022-08-31:
After I have tested the "simple-viz --pcap OS1-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap --meta OS1-128_Rev-06_fw23_Urban-Drive_Dual-Returns.json",  the display will visualize the lidar data .

But run the following commadn,
python3 -m ouster.sdk.examples.pcap OS1-128_Rev-06_fw23_Urban-Drive_Dual-Returns.pcap OS1-128_Rev-06_fw23_Urban-Drive_Dual-Returns.json read-packets,
there is no show on my conda environment.

oneway3124, 2022-08-31:
Similar phenomenon with "python3 -m ouster.sdk.examples.open3d \
    --pcap $SAMPLE_DATA_PCAP_PATH --meta $SAMPLE_DATA_JSON_PATH --start 84 --pause"

kairenw, 2022-08-31:
What's the `lidar_port` on your sensor? You can find it on your sensor homepage: http://os-xxxxxxxxxxxx.local

`simple-viz` (a utility) will use the lidar port your sensor is already configured to but the python examples expect the ports 7502 and 7503 for lidar and imu data respectively.

kairenw, 2022-09-15:
Closing as there is no response, and the ports were likely the issue

closed 2022-09-15 by kairenw