ouster-lidar/ouster-sdk #427: can not get data from code

JohnHe12, 2022-08-31:
i use the ouster lidar and use the code from github, and from the code it shows nothing and the messages in the topic is empty.
so i use the wireshark to save pcap ,use open3d to viz, it shows the pointcloud ,so it send the data to the port. and webgui has no error. and then i use my laptop under the same internet, it works! so same internet, same code , same lidar ,of course i check the different ip from computer , but my computer can not viz nor publish. it very confused me.

Samahu, 2022-08-31:
Hi @JohnHe12, are you using the latest version of the example, can you share the logs from the ROS launch command?

JohnHe12, 2022-08-31:
@Samahu ,because i use the FW 2.1.3 so i use the corresponding example 20210608 and the launch file on the laptop and computer is the same.follow is my launch file. what confused me is that the same environment in laptop and computer . last week my computer could receive the data, after i set up the NTP and change the GPU, it didn't work until now.
[image]

Samahu, 2022-08-31:
You could still upgrade to the last SDK release (if you wish). The last software release named FW 2.4 supports FW 2.0 and higher. Upgrading may or may not help with your case.

I asked for the output of the launch command, meaning what is the console output after you execute `roslaunch ouster_ros ...`.

Since you mentioned that the same code works on a different machine, I feel the problem you are dealing with is specific to your computer/sensor/network setup rather than a problem with the code. You may want to reach to [our customer support team](https://ouster.atlassian.net/servicedesk/customer/portal/8) for this kind of issues.

JohnHe12, 2022-08-31:
@Samahu Yeah i totally agree with you , and i have already read the document and i don't find the solution.because i can ping the lidar.

kairenw, 2022-08-31:
Is it possible you've got a firewall on?

JohnHe12, 2022-08-31:
@kairenw ufw is on, you mean i need to turn off firewall first?

kairenw, 2022-09-01:
yes that's correct (and restart!)

JohnHe12, 2022-09-02:
@kairenw i sove the problem , it's the firewall problem!!!!! thanks very much.

kairenw, 2022-09-02:
Oh, great! I'm so glad. Sorry I didn't get back to you in time to confirm. Glad it got fixed.

Closing this ticket now

closed 2022-09-02 by kairenw