ouster-lidar/ouster-sdk #428: a lot of data is lost in the recorded rosbag

txyxy, 2022-09-02:
Hi, we recorded the messages published from the ros node. After recording we checked the rosbag and found there are a lot of data lost. We checked the ethernet port transmission rate, it's 1000Mb/s. We used another network cable and got the same result. Could you give me some suggestions to fix this issue?

[image]

Samahu, 2022-09-02:
This definitely looks unhealthy recording. Here is what typically things should look like:
[image]

* Are you connected directly to the sensor or via a switch?
* Do you have other devices that might be transmitting data to same machine
* In case you have a firewall, disable temporarily and re-try the recording

Samahu, 2022-09-02:
Two other thoughts:

* What firmware is your sensor on? consider upgrading to latest in case you are not.
* Check for any active alerts sensor from the sensor diagnostics page

Samahu, 2022-09-15:
No response so far, closing this issue. Let us know if you are still having issues

closed 2022-09-15 by Samahu

txyxy, 2022-11-03:
Thank you for your professional advicese and sorry to reply you so late. We still have problems with recording ros messages. For example, after we extracted msgs from the recorded imu rosbag, we found some thing strage like below:
[image]
[image]

The timestamp of messages are confused. For example, message 19147 is sandwiched between messages 19127 and 19128. We  gusse that 19147 is a later one but got an advanced timestamp. We also gusse that 19147 is one of those "lost messages". 

We've been obsessed with this question for a long time so we plan to record in a different format. For recording data with unlimited duration, do you recommend using the python sdk to record data in pcap format? I see an example of pyhton, but it can only record 10 seconds of pcap data.

Samahu, 2022-11-03:
You can increase the duration of recorded at this line:
https://github.com/ouster-lidar/ouster_example/blob/fa045e561600e4fd4227d0c95283f0d27e26f1e0/python/src/ouster/sdk/examples/client.py#L234

You can also modify the sample code to record until a certain condition is met.

txyxy, 2022-11-04:
OK, I'll try it. Thanks!

Samahu, 2022-11-04:
> OK, I'll try it. Thanks!

You are welcome.

Sorry I almost missed the other part of your questions

>The timestamp of messages are confused. For example, message 19147 is sandwiched between messages 19127 and 19128. We gusse that 19147 is a later one but got an advanced timestamp. We also gusse that 19147 is one of those "lost messages".

We rely on UDP to transfer IMU messages. Unlike TPC, UDP does not guarantee delivery or order of delivered messages. So messages coming out of order is expected. It should be possible however to write some logic to re-order messages by time or sequence number (within a specific window of time) or drop packets that are older than the last one received. We do some of that when assemble the lidar scan frame. Hope this helps!