ouster-lidar/ouster-sdk #429: Ghost point in the cloud after new update

shubham-shahh, 2022-09-05:
Hi, 
I am getting a ghost point right in the center of the point cloud and it stays
I have attached an image below which shows the point

thanks
[image]

kairenw, 2022-09-06:
Hi @shubham-shahh, 

That point has always been there, I believe. It's an artifact of how we're doing the XYZ projection. (0 in the range gets mapped to 0,0,0)
Is it causing significant issues for you?

shubham-shahh, 2022-09-06:
Hi @kairenw, thanks for the response, In my case, the lidar is mounted on a drone and I have included a euclidian distance filter (equal to my drone's radius) [here](https://github.com/ouster-lidar/ouster_example/blob/da9ba6fb300119c222db17b6814dfffc2999f774/ouster_ros/src/ros.cpp#L149) to make sure, propellers don't come in the cloud field, but even after applying the filter, this point still apprears 

thanks

kairenw, 2022-09-06:
Hi @shubham-shahh,

How is your filter implemented? The thing is that at the beginning of that function when [resize is called](https://github.com/ouster-lidar/ouster_example/blob/da9ba6fb300119c222db17b6814dfffc2999f774/ouster_ros/src/ros.cpp#L127), the cloud is filled with 0 points already. So even if you filter what gets written to the point cloud again, you're going to get the ghost point. What you probably want to do is use an unstructured point cloud instead and `push_back` every time you have a nonzero-range point.

shubham-shahh, 2022-09-07:
Hi @kairenw 

I have implemented the filter here

```
for (auto u = 0; u < ls.h; u++) {
        for (auto v = 0; v < ls.w; v++) {
            const auto xyz = points.row(u * ls.w + v);
            const auto ts =
                (std::chrono::nanoseconds(timestamp[v]) - scan_ts).count();

                float eucledian_dist = sqrt(pow(static_cast<float>(xyz(0)) - 0.0f, 2) +
                            pow(static_cast<float>(xyz(1)) - 0.0f, 2) +
                            pow(static_cast<float>(xyz(2)) - 0.0f, 2) * 1.0);
           if(eucledian_dist> drone_radius){
            cloud(v, u) = ouster_ros::Point{
                {{static_cast<float>(xyz(0)), static_cast<float>(xyz(1)),
                  static_cast<float>(xyz(2)), 1.0f}},
                static_cast<float>(signal(u, v)),
                static_cast<uint32_t>(ts),
                static_cast<uint16_t>(reflectivity(u, v)),
                static_cast<uint8_t>(u),
                static_cast<uint16_t>(near_ir(u, v)),
                static_cast<uint32_t>(range(u, v))};
            }
        }
    }
```
I have noted your suggestion and I'll try that

Also, I am facing this problem, if I fly the drone at 10m height in a complete obstacle-less environment and using the euclidean filter to filter the propeller points. I am getting points just outside the drone_radius (distance from lidar center to propeller tip) and In my understanding, there are no points there. according to you, what could be the possible reason for that? does the cage design (in the blind zone of lidar) create any issues

thanks

kairenw, 2022-09-07:
Yes, this filter is exactly what I was saying -- you're filtering on the distance but there's already a 0 point at that (u,v) due to the `resize`.

As for your other question -- I think it'd probably be easier if you showed us some photos etc on a ticket in [Customer Support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) (as this is getting a beyond use of the SDK). If you can provide a sample of what the data looks like, with the poitns you're concerned about circled, and also a photo of the setup, that will probably help a lot.

kairenw, 2022-09-07:
I'm going to close this ticket now, but feel free to reopen it if you have more questions.

closed 2022-09-07 by kairenw

Samahu, 2022-09-07:
@shubham-shahh as an alternative solution you can actually keep `ros.cpp` unchanged and then apply a PCL PassThrough filter or any distance based filter on the produced point cloud right before it gets published in cloud node.

shubham-shahh, 2022-09-08:
> @shubham-shahh as an alternative solution you can actually keep `ros.cpp` unchanged and then apply a PCL **PassThrough** filter or any distance based filter on the produced point cloud right before it gets published in cloud node.

Hi @Samahu,
Thanks for the response. I will try your suggestion. I was hesitant to add a filter after the point cloud is made to avoid any processing latency it may introduce