ouster-lidar/ouster-sdk #433: Question about ouster lidar property

bigbigpark, 2022-09-20:
Hi, I wonder where is the code that computes `reflectivity`, `noise` and `ring`. <br/>

Could you tell me which `.cpp` file doing this job? <br/>

Thanks.

bigbigpark, 2022-09-20:
I solved this problem.

closed 2022-09-20 by bigbigpark

juanluishortelano, 2023-02-14:
Could you write here where did you find them for future reference? Thanks

bigbigpark, 2023-02-14:
Hi, you can find them by running ROS example with a bag file.

Point cloud from Ouster lidar has the mentioned properties.

I hope this will be helpful.

juanluishortelano, 2023-02-14:
It is helpful but did you find the .cpp file that computes reflectivity?

Samahu, 2023-02-14:
@juanluishortelano the SDK does not compute reflectivity, it comes directly from the sensor. You can refer to [this previous issue](https://github.com/ouster-lidar/ouster_example/issues/412) which mentions given an approximate answer to how reflectivity is computed.

juanluishortelano, 2023-02-16:
Thanks, that is very interesting. My question was actually spawned by reading your answer in that issue. 

My confusion comes from the fact that intensity is the actual measurement that the sensor takes, if reflectivity also comes from the sensor through UDP: does that mean that the calculation reflectivity = cte * intensity * range^2 is done in the sensor itself? Does it not send the raw intensity value then?

Samahu, 2023-02-16:
It does, it is basically what we now refer to as the SIGNAL.

juanluishortelano, 2023-02-17:
Great, thank you for your answers!