ouster-lidar/ouster-sdk #435: Simulate lower beam on Ouster OS2-128

vebjornjr, 2022-09-22:
Hello.

I have a Ouster OS2-128 sensor and I want to see how my processing performs on a sensor with lower number of beams.

What is the easiest way to get my sensor to behave as a 64 or 32 beam?

Thank you!

kairenw, 2022-09-22:
I'd try talking to our [customer support]( https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) about this.

You can also try running your system [sample data](https://ouster.com/resources/lidar-sample-data/) that has the desired number of beams / `lidar_mode` / lidar profile. I recall you're using the C++ SDK directly (with no ROS)? You can use `[ouster_pcap](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_pcap/include/ouster/os_pcap.h)` to replay packets as needed.

vebjornjr, 2022-09-27:
C++ SDK right. The sample data is not relevant for our case, so I would really like to be able to emulate lower beam with your SDK. Is there not any way possible to hack with the metadata/sensor info to achieve it? I don't need any automatic way, if I can achieve it by manually editing some jsons/pcap then it works.

kairenw, 2022-09-27:
Can you come talk to us via [customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) about your use case? Thanks!

closed 2022-09-27 by kairenw