ouster-lidar/ouster-sdk #437: Issue with init_client() in 0.5.1

stockerd, 2022-09-26:
In our VS2019/CPP application, we use the long-form prototype of init_client(), primarily because we set the timeout value.  We also set the lidar_mode parameter in that call.  This worked in 0.5.0, but we get a runtime error when running with 0.5.1.

We have duplicated this by modifying client_example.exe in the CMake project, which generates the same error:

//  auto handle = sensor::init_client(sensor_hostname, data_destination);	// ORIGINAL CALL
    auto handle = sensor::init_client(sensor_hostname, data_destination, ouster::sensor::lidar_mode::MODE_512x10);	  // GENERATES ERROR BELOW

>client_example.exe 192.168.1.8
Ouster client example 0.5.1
Connecting to "192.168.1.8"... init_client error: CurlClient::execute_get failed for url: [http://192.168.1.8/api/v1/sensor/cmd/set_config_param?args=lidar_mode+1] with the code: [500] - and return: {"error": {"title": "500 Internal Server Error", "description": "Command \"set_config_param lidar_mode 1\" failed: error: Invalid configuration value"}}
Failed to connect

The target sensor is a new (Gen 6) OS-0-32-U0 (Gen 6) with v2.3.1 firmware.

This may be a "pilot error" on our part (e.g., the wrong version of curl? - although we just pulled and generated it last week), but it's not clear.

Samahu, 2022-09-26:
This seems like a bug on our side, I would suggest as a work around to use:
```C++
auto handle = sensor::init_client(sensor_hostname, data_destination);
```
followed by
```C++
sensor::sensor_config config;
lidar_mode = sensor::lidar_mode_of_string(lidar_mode_arg);
config.ld_mode = lidar_mode;
sensor::set_config(hostname, config, 0);
```

stockerd, 2022-09-26:
Thanks for the quick response.

We configure the sensor in the workflow of our app drive anyway, so we already had an effective workaround for setting lidar_mode.  We have maintained the longer prototype for init_client(), but now (at least temporarily) set lidar_mode and ts_mode to *UNSPEC, so the calls to sensor_http->set_config_param() within init_client() are skipped.

Samahu, 2022-09-28:
@stockerd This PR #438 should address the bug you have encountered. Closing this ticket.

closed 2022-09-28 by Samahu