ouster-lidar/ouster-sdk #439: sensor.info.status and telemetry data support in C++ SDK

stockerd, 2022-09-29:
What is the plan for the implementation of the new status field into the sensor_info struct and the new telemetry data for the C++ SDK?

The status field is in the critical path for us to deploy any Ouster sensors in production - we must be able to inform our automation system and users of a cold start delay in our outdoor applications.

kairenw, 2022-09-29:
Hi @stockerd,

Thanks for reaching out. I understand you're talking to one of our FAEs about these requests. I've asked him to send you back some questions! I'll keep this ticket open, but let's try to talk through that channel where possible.

Thanks!

stockerd, 2022-09-30:
Understand - thanks.

kairenw, 2022-11-08:
Hi @stockerd, I managed to chat to the FAE on our account recently and he mentioned that most of your issues were resolved. Is this true? Let us know!

stockerd, 2022-11-09:
Yes, most of our issues have been resolved.  But we appreciate the opportunity to consult on the best practices you guys would recommend for integrating your C++ SDK into our application suite.  Also, we are looking for the sensor status to be presented in the sensor_info struct in the C++ SDK.

Micheal T has my direct contact info.  Thanks

kairenw, 2022-11-10:
Hey @stockerd, 

We're unlikely to introduce the `status` into the C++ `sensor_info` struct currently (despite the unfortunate name collision, the C++ `sensor_info` is not meant to be a direct representation of the `sensor_info` firmware concept), but we are interested in how you'd like to use the suggested feature (so we can find some other way to meet your need). Feel free to write here or through Michael, thanks!

stockerd, 2022-12-01:
@kairenw:  We are primarily looking for the status of the sensor for abnormal cases; one example is to be aware when sensor start-up is delayed during warm-up in cold conditions.

Based on experience applying LIDAR in our systems over the past 25+ years, we find this type of sensor status information and also telemetry is critical for supporting robust operation, efficient maintenance, and minimum downtown for our end-users.

And all of our relevant software applications are coded primarily in C++.

kairenw, 2022-12-02:
That makes sense. 

So, the way the client is written currently, the C++ `sensor_info` struct is only populated once -- when we first pull the metadata while initializing. Then it never gets updated again. So even if we introduced `status` into the `sensor_info` struct,  it wouldn't reflect any changes as the sensor continues running,. though I guess it could specifically solve the sensor start-up delay situation you mention.

When you say telemetry, are you referring to anything in particular?

It's definitely not ideal that the SDK is essentially is missing a layer of support, both in providing the `status` from the FW `sensor_info` and also in lacking an alert interface. I can't promise a solution or new feature within the next few months, however, though the team has noted the feature request.

I'd suggest that in the meantime you use the http endpoint to poll independently, and just parse the json for the `status`.