ouster-lidar/ouster-sdk #441: Failed to initialize sensor at: 192.168.x.x [ROS]

aristow1, 2022-10-13:
Hi everyone,
so im trying to use the OS1-32 with ros and when i launch the os1.launch file i get this error :
init_client: command "set_udp_ip 192.168.0.10" failed with "error: Command not recognized."
Failed to initialize sensor at: 192.168.2.10
[ouster_driver-1] process has died [pid 23981, exit code 1, cmd /home/heros/simulation_ws/devel/lib/ouster_driver/ouster_driver __name:=ouster_driver __log:=/home/heros/.ros/log/5ebe4d9e-4adb-11ed-9728-00187da3b2d3/ouster_driver-1.log].
log file: /home/heros/.ros/log/5ebe4d9e-4adb-11ed-9728-00187da3b2d3/ouster_driver-1*.log

the sensor is connected to a eth0 interface with the ip adress : 192.168.0.10
and the sensor has an IP adress of : 192.168.2.10.
im kinda new to all this so im hoping to get some help on how to solve this problem.
Im using Ubuntu 16.04 and ros kinetic

kairenw, 2022-10-13:
Hi @aristow1, 

I'm not really sure when your code is from. `set_udp_ip` is a command that we deprecated and removed from the sensor firmware two years ago, so my guess is that your version of the client is at least two years old.

It would also be helpful if you tell us which FW is on your sensor. You should be able to find this out by visiting the sensor homepage, or the IP address assigned to the sensor.

ROS kinetic is no longer supported by the official Ouster ROS1 driver, now located here: [ouster-ros](https://github.com/ouster-lidar/ouster-ros). However, you can still get started with Ubuntu 18.04 and ROS melodic with the instructions at that repo!

aristow1, 2022-10-14:
@kairenw i just downloaded the ouster software on my other ubuntu 20.04 PC to try and see the OS1 configuration but when scan the port in which the sensor is connected to it cant seem to detect the sensor and when i give the IP in the software manually the software doesnt respond. however when i arp-scan the serial port in which the sensor is connected to in the PC i get an IP-adress of the sensor.

aristow1, 2022-10-14:
@kairenw i just unplugged and replugged the sensor and i managed to ping it : and this is the FW version :     ousteros-image-prod-aries-v2.0.0

aristow1, 2022-10-14:
@kairenw  so i updated my FW to v2.4 and when i tried testing the sensor with OUSTER software and try scanning for the sensor the software cant seem to find it. and when i type the IP-Adress of the sensor manually the configuration window pops out but i cant seem to click the OK button to start the sensor stream

aristow1, 2022-10-14:
@kairenw  after succesfully managing to ping the ouster sensor and updating the FW verison to v2.4, the ouster studio software still cant find the sensor automatically when i try to scan for it. however when i type in the IP adress of the sensor manually, i manage to to do all the configuration, but when i press on the "OK" button, i dont get any sensor data shown. and after trying it with ROS again i get this : 

init_client: command "set_udp_port_lidar 7502" failed with "error: Command not recognized."
init_client: command "set_udp_port_imu 7503" failed with "error: Command not recognized."
init_client: command "set_udp_ip 192.168.0.10" failed with "error: Command not recognized." ==> this is the eth0 interface where the sensor is plugged into.
[ERROR] [1665745239.870300330]: Failed to initialize sensor at: os-122xxxxxxxxx.local
[ouster_driver-1] process has died [pid 7460, exit code 1, cmd /home/heros/simulation_ws/devel/lib/ouster_driver/ouster_driver __name:=ouster_driver __log:=/home/heros/.ros/log/be2d8f46-4bae-11ed-84fb-00187da3b2d3/ouster_driver-1.log].
log file: /home/heros/.ros/log/be2d8f46-4bae-11ed-84fb-00187da3b2d3/ouster_driver-1*.log

i Hope i can get help on this matter. Thx

kairenw, 2022-10-14:
Did you update the code you're running? That code is really outdated and also is not maintained by this repo's owners.

aristow1, 2022-10-17:
after using the official ouster-ros repo, and launching the sensor.launch file. i still cant get Pointcloud data( i cant seen anything in Rviz). this is the terminal output after launch : 
 INFO] [1666083023.530908282]: Initializing nodelet with 16 worker threads.
[ INFO] [1666083024.515276759]: Loading nodelet /ouster/os_node of type nodelets_os/OusterSensor to manager os_nodelet_mgr with the following remappings:
[ INFO] [1666083024.542085604]: Will send UDP data to 192.168.x.x
[ WARN] [1666083024.628711499]: Sensor 192.168.x.x configured successfully
[ INFO] [1666083024.628845933]: Starting sensor 192.168.x.x initialization...
[ INFO] [1666083025.522207336]: Loading nodelet /ouster/os_cloud_node of type nodelets_os/OusterCloud to manager os_nodelet_mgr with the following remappings:
[ INFO] [1666083025.527204792]: Loading nodelet /ouster/img_node of type nodelets_os/OusterImage to manager os_nodelet_mgr with the following remappings:
[ INFO] [1666083026.737842907]: rviz version 1.14.19
[ INFO] [1666083026.737917549]: compiled against Qt version 5.12.8
[ INFO] [1666083026.737944673]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1666083026.749686098]: Forcing OpenGl version 0.
[ INFO] [1666083027.041008587]: Stereo is NOT SUPPORTED
[ INFO] [1666083027.041068613]: OpenGL device: Mesa Intel(R) UHD Graphics 630 (CML GT2)
[ INFO] [1666083027.041104565]: OpenGl version: 4,6 (GLSL 4,6) limited to GLSL 1.4 on Mesa system.
[ WARN] [1666083038.671769716]: Client version: 0.5.2+a12011a-release
[ WARN] [1666083038.671862569]: Using lidar_mode: 1024x10
[ WARN] [1666083038.671902791]: OS-1-32-U0 sn: 122108000456 firmware rev: v2.4.0
[ WARN] [1666083038.674092626]: No metadata file was specified, using: 192.168.0-metadata.json
[ INFO] [1666083038.674400141]: Wrote metadata to 192.168.0-metadata.json
[ INFO] [1666083038.675790493]: get_metadata service created
[ INFO] [1666083038.677081715]: get_config service created
[ INFO] [1666083038.678359707]: set_config service created
[ INFO] [1666083039.682948237]: OusterCloud: retrieved sensor metadata!
[ INFO] [1666083039.684048674]: Profile has 1 return(s)
[ INFO] [1666083040.684347913]: OusterImage: retrieved sensor metadata!

i updated the ip of the sensor and the interface in the launch file.
i also cant get lidar data with the official ouster software.
any idea on what the problem here might be ?

kairenw, 2022-10-17:
Can I see your launch command?

aristow1, 2022-10-17:
roslaunch sensor.launch
... logging to /home/xxxx/.ros/log/e58fd880-4ec1-11ed-9290-fd598655a95a/roslaunch-xxxx-AMB-5210G-5605.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xxxx-AMB-5210G:46479/

SUMMARY
========

PARAMETERS
 * /ouster/os_cloud_node/tf_prefix: 
 * /ouster/os_cloud_node/timestamp_mode: 
 * /ouster/os_node/imu_port: 0
 * /ouster/os_node/lidar_mode: 
 * /ouster/os_node/lidar_port: 0
 * /ouster/os_node/metadata: 
 * /ouster/os_node/sensor_hostname: 192.168.0.3
 * /ouster/os_node/timestamp_mode: 
 * /ouster/os_node/udp_dest: 192.168.0.111
 * /ouster/os_node/udp_profile_lidar: 
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    relay_imu (topic_tools/relay)
    relay_lidar (topic_tools/relay)
    rviz (rviz/rviz)
  /ouster/
    img_node (nodelet/nodelet)
    os_cloud_node (nodelet/nodelet)
    os_node (nodelet/nodelet)
    os_nodelet_mgr (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5628]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e58fd880-4ec1-11ed-9290-fd598655a95a
process[rosout-1]: started with pid [5653]
started core service [/rosout]
process[ouster/os_nodelet_mgr-2]: started with pid [5660]
process[ouster/os_node-3]: started with pid [5662]
process[ouster/os_cloud_node-4]: started with pid [5664]
process[ouster/img_node-5]: started with pid [5666]
process[rviz-6]: started with pid [5668]
process[relay_lidar-7]: started with pid [5670]
process[relay_imu-8]: started with pid [5671]
[ INFO] [1666083023.530908282]: Initializing nodelet with 16 worker threads.
[ INFO] [1666083024.515276759]: Loading nodelet /ouster/os_node of type nodelets_os/OusterSensor to manager os_nodelet_mgr with the following remappings:
[ INFO] [1666083024.542085604]: Will send UDP data to 192.168.x.xxx
[ WARN] [1666083024.628711499]: Sensor 192.168.0.3 configured successfully
[ INFO] [1666083024.628845933]: Starting sensor 192.168.x.xxx initialization...
[ INFO] [1666083025.522207336]: Loading nodelet /ouster/os_cloud_node of type nodelets_os/OusterCloud to manager os_nodelet_mgr with the following remappings:
[ INFO] [1666083025.527204792]: Loading nodelet /ouster/img_node of type nodelets_os/OusterImage to manager os_nodelet_mgr with the following remappings:
[ INFO] [1666083026.737842907]: rviz version 1.14.19
[ INFO] [1666083026.737917549]: compiled against Qt version 5.12.8
[ INFO] [1666083026.737944673]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1666083026.749686098]: Forcing OpenGl version 0.
[ INFO] [1666083027.041008587]: Stereo is NOT SUPPORTED
[ INFO] [1666083027.041068613]: OpenGL device: Mesa Intel(R) UHD Graphics 630 (CML GT2)
[ INFO] [1666083027.041104565]: OpenGl version: 4,6 (GLSL 4,6) limited to GLSL 1.4 on Mesa system.
[ WARN] [1666083038.671769716]: Client version: 0.5.2+a12011a-release
[ WARN] [1666083038.671862569]: Using lidar_mode: 1024x10
[ WARN] [1666083038.671902791]: OS-1-32-U0 sn: 12xxxxxxxxxx firmware rev: v2.4.0
[ WARN] [1666083038.674092626]: No metadata file was specified, using: 192.168.0-metadata.json
[ INFO] [1666083038.674400141]: Wrote metadata to 192.168.x-metadata.json
[ INFO] [1666083038.675790493]: get_metadata service created
[ INFO] [1666083038.677081715]: get_config service created
[ INFO] [1666083038.678359707]: set_config service created
[ INFO] [1666083039.682948237]: OusterCloud: retrieved sensor metadata!
[ INFO] [1666083039.684048674]: Profile has 1 return(s)
[ INFO] [1666083040.684347913]: OusterImage: retrieved sensor metadata!

aristow1, 2022-10-17:
@kairenw i managed to solve the Problem by switching to an Ubuntu 18.04 , Ros Melodic PC

closed 2022-10-19 by aristow1