ouster-lidar/ouster-sdk #444: Read certain range of packets repeatedly from a pcap file

dasol92, 2022-10-21:
Hello, experts
- I'm working with ouster_example v20220927.

I faced an issue that read certain range of packets repeatedly from a pcap file.
`Tins::Packet` should reads next packet sequentially, but it came back to first frame of the pcap file
`Tins::Packet` should read the next packet sequentially, but after reading a few packets, it came back to the first frame of the pcap file, so it goes on over again and again certain range of packets.

I'm working with the `examples/helpers` function.
from `examples/helpers.cpp`
```cpp
while (ouster::sensor_utils::next_packet_info(*handle, packet_info)) {  // @dasol: Ready for the next packet 
        auto packet_size = ouster::sensor_utils::read_packet(           // @dasol: Read once a packet
            *handle, packet_buf.get(), BUF_SIZE);
        ...
}
```

How can i solve it?
Thanks.

kairenw, 2022-10-21:
Hi @dasol92,

That seems strange! Mind submitting your pcap through our customer support [here](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86)?

We can take a look at it on our side

dasol92, 2022-10-23:
I attached pcap&json to the customer support. Could you check the files?
Thanks.

dasol92, 2022-10-24:
Hi @kairenw,
That works! (BTW it's too fast)
I'd comment line 20 not to check udp lidar port.
```cpp
packet_info.dst_port == info.udp_port_lidar) { ...
````

As i mentioned above, by the way, the playback speed is too fast.
I can't find parsing timestamp or sleep code (for sync with sensor real-time) in sdk C++ examples.
However, Python SDK viz example displays at the recorded speed (recorded with Ouster Studio).

How do I play the recorded speed with the C++ codes?

Thanks a lot!

kairenw, 2022-10-24:
Hi @dasol92,

If you want to play it back at recorded speed, you'll have to handle the timestamps yourself. You can use the timestamp in `packet_info.timestamp` and `std::chrono` to keep track of the current time. The logic that is used [here](https://github.com/ouster-lidar/ouster_example/blob/master/python/src/ouster/pcap/pcap.py#L218) in the Python iterator for pcap is basically what you want, but to duplicate it into C++.

Hope that helps!

kairenw, 2022-11-09:
Closing this out. Thanks! Let us know if you need anything else.

closed 2022-11-09 by kairenw