ouster-lidar/ouster-sdk #445: Proprietary vs BSD License

monidzik, 2022-10-24:
@kairenw In the cpp files in the repository I see the proprietary license:
 ```
Copyright (c) 2018, Ouster, Inc.
All rights reserved.
``` 
while the LICENSE file in the top of hierarchy mentions BSD license:
```
Copyright: 2018-2022 Ouster, Inc
License: BSD-3-Clause
```

Could you clarify which license applies?

kairenw, 2022-11-08:
To clarify, "All rights reserved" means the copyright holder (Ouster, in this case) reserves all rights unless it grants others express permissions.  The BSD license is an express permission from Ouster to others governing the use of the code, and any rights not explicitly granted under the BSD license are still reserved by Ouster. 

Hope that helps!

closed 2022-11-08 by kairenw

monidzik, 2022-11-09:
Thanks @kairenw for the answer and explanation, it's clear now; but I also think it would be great if you correct the source of ambiguity by placing a consistent license header in all files.