ouster-lidar/ouster-sdk #447: Data release exception in client_example.cpp

zpy412898523, 2022-10-31:
Variables in 142 lines of code    ”std::vector<LidarScan::Points> clouds“    
When the program runs out of the method body scope, the memory release exception will occur when the variable is released

Samahu, 2022-10-31:
Hi @zpy412898523, can you please share the type of memory exception that you have observed? and how you produced the issue?

zpy412898523, 2022-11-02:
DWORD __stdcall CDeviceCollection::s_ThreadLidar(LPVOID lpParam)
{
	LIDAR_INFO* lidar = (LIDAR_INFO*)lpParam;
	CDeviceCollection* pThis = (CDeviceCollection*)g_pDeviveCollection;

	while (!lidar->beLidarOpen.load())
	{
		Sleep(1000);
	}

	/*
	* Configuration and calibration parameters can be queried directly from the
	* sensor. These are required for parsing the packet stream and calculating
	* accurate point clouds.
	*/
	std::cerr << "Gathering metadata..." << std::endl;
	auto metadata = sensor::get_metadata(*lidar->hClient);

	// Raw metadata can be parsed into a `sensor_info` struct
	sensor::sensor_info info = sensor::parse_metadata(metadata);

	size_t w = info.format.columns_per_frame;
	size_t h = info.format.pixels_per_column;

	ouster::sensor::ColumnWindow column_window = info.format.column_window;

	std::cerr << "  Firmware version:  " << info.fw_rev
		<< "\n  Serial number:     " << info.sn
		<< "\n  Product line:      " << info.prod_line
		<< "\n  Scan dimensions:   " << w << " x " << h
		<< "\n  Column window:     [" << column_window.first << ", "
		<< column_window.second << "]" << std::endl;

	// A LidarScan holds lidar data for an entire rotation of the device
	std::vector<LidarScan> scans{
		N_SCANS, LidarScan{w, h, info.format.udp_profile_lidar} };

	// A ScanBatcher can be used to batch packets into scans
	sensor::packet_format pf = sensor::get_format(info);
	ScanBatcher batch_to_scan(info.format.columns_per_frame, pf);

	/*
	 * The network client provides some convenience wrappers around socket APIs
	 * to facilitate reading lidar and IMU data from the network. It is also
	 * possible to configure the sensor offline and read data directly from a
	 * UDP socket.
	 */
	std::cerr << "Capturing points... ";

	// buffer to store raw packet data
	auto packet_buf = std::make_unique<uint8_t[]>(UDP_BUF_SIZE);

	XYZLut lut = make_xyz_lut(info);

	while (true)
	{
		if (!lidar->beGrab.load())
		{
			Sleep(1000);
			continue;
		}

		for (size_t i = 0; i < N_SCANS;) {
			// wait until sensor data is available
			sensor::client_state st = sensor::poll_client(*lidar->hClient);
			//count++;

			// check for error status
			if (st & sensor::CLIENT_ERROR)
			{
				LOG(INFO) << "Sensor client returned error state!";
				break;
			}

			// check for lidar data, read a packet and add it to the current batch
			if (st & sensor::LIDAR_DATA) {

				if (!sensor::read_lidar_packet(*lidar->hClient, packet_buf.get(), pf)) {
					LOG(INFO) << "Failed to read a packet of the expected size!";
				}

				// batcher will return "true" when the current scan is complete扫描完成后，批处理程序将返回“true”
				if (batch_to_scan(packet_buf.get(), scans[i])) {
					// retry until we receive a full set of valid measurements
					// (accounting for azimuth_window settings if any)
					//重试，直到我们收到一整套有效的测量值
					//（考虑方位角窗口设置（如果有））

					if (scans[i].complete(info.format.column_window)) {
						
						auto cloud = cartesian(scans[i] , lut);
						pThis->m_cbLidarData(cloud, scans[i].field(ouster::sensor::SIGNAL).data(), lidar);
						i++;
					}
				}
			}

			if (st & sensor::IMU_DATA) {
				sensor::read_imu_packet(*lidar->hClient, packet_buf.get(), pf);
			}
		}

	}

	return 0;
}

An exception occurs when the cloud variable of the data collection thread code of the above ouster Lidar is released

zpy412898523, 2022-11-02:
[image]

zpy412898523, 2022-11-02:
[image]

zpy412898523, 2022-11-02:
[image]

zpy412898523, 2022-11-02:
An exception will also be thrown when setting the lidar config parameters
[image]

zpy412898523, 2022-11-02:
windows x64 vs2019

Samahu, 2022-11-03:
I see you have integrated our code into another program. Are you able to produce the same issue merely using our example code?

kairenw, 2023-07-07:
Closed for no response - please reopen if you still desire support

closed 2023-07-07 by kairenw