ouster-lidar/ouster-sdk #450: Recorded pcap using Ouster studio does not work with the cpp pcap_replay

AndjiAgileo, 2022-11-08:
Hey,

I've been trying to implement a ouster sensor in my project and right now I'm implementing the Pcap reading part. When I record a pcap file using the CPP SDK functions (ouster::sensor_utils::record_initialize and stuff), the playback of the data works (the scans are complete based on the ScanBatcher). But if I try to read a pcap file that comes from ouster studio (using the record function), the ScanBatcher refuses to give me a complete scan (the LidarScan::complete method always returns false).

It's really a bummer, as using OusterStudio to record pcap files is much easier and convenient.

kairenw, 2022-11-08:
So what `complete` does is it checks the LidarScan `status` to make sure every column is present. Have you taken a look at the output there to see if it's complete? If it isn't, it's likely you have dropped packets when recording from Ouster Studio.

You can use the Python SDK to record data if you like -- you may find that a bit easier. Here's an [example] (https://static.ouster.dev/sdk-docs/python/examples/record-stream.html#recording-sensor-data)

kairenw, 2022-11-09:
Can you show me how you're playing it back? I don't recommend using `complete()` to decide whether or not to accept a batched scan from `ScanBatcher`

AndjiA2, 2022-11-10:
In the I managed to solve my issue, it was mostly me not quite understanding how the pcap reading functions worked... Now I'm able to properly read the pcap recordings from OusterStudio via c++ code.

closed 2022-11-10 by AndjiA2